ZoneMinder/zoneminder #2292: Cant start zoneminder.service - Unable connect to zmdc.sock

Frankstar, 2018-11-02:
As the Title says, i cant start the zoneminder.service.
The weird thing about it - it work after install.
i stoped the service via the webconsole - and now i cant start it again.
(not from webconsole, not from terminal

**ZoneMinder Version (`zmaudit.pl -v`):**
1.32.2

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
NAME=Fedora
VERSION="29 (Server Edition)"

Installed via rpmfusion

**Relevant log lines:**
```
Nov 02 11:52:28 localhost.localdomain systemd[1]: Starting ZoneMinder CCTV recording and security system...
Nov 02 11:52:31 localhost.localdomain zmpkg[1225]: ERR [Unable to run "/usr/bin/zmdc.pl start zmfilter.pl --filter_id=2 --daemon", output is "Unable to connect to server using socket at /var/lib/zoneminder/sock/z>
Nov 02 11:52:32 localhost.localdomain zmpkg[1225]: ERR [Unable to run "/usr/bin/zmdc.pl start zmaudit.pl -c", output is "Unable to connect to server using socket at /var/lib/zoneminder/sock/zmdc.sock", status is >
Nov 02 11:52:32 localhost.localdomain zmpkg[1225]: ERR [Unable to run "/usr/bin/zmdc.pl start zmwatch.pl", output is "Unable to connect to server using socket at /var/lib/zoneminder/sock/zmdc.sock", status is 255]
Nov 02 11:52:33 localhost.localdomain zmpkg[1225]: ERR [Unable to run "/usr/bin/zmdc.pl start zmtelemetry.pl", output is "Unable to connect to server using socket at /var/lib/zoneminder/sock/zmdc.sock", status is>
Nov 02 11:52:33 localhost.localdomain zmpkg[1225]: ERR [Unable to run "/usr/bin/zmdc.pl start zmstats.pl", output is "Unable to connect to server using socket at /var/lib/zoneminder/sock/zmdc.sock", status is 255]
Nov 02 11:52:33 localhost.localdomain systemd[1]: zoneminder.service: New main PID 1233 does not exist or is a zombie.
Nov 02 11:53:58 localhost.localdomain systemd[1]: zoneminder.service: Start operation timed out. Terminating.
Nov 02 11:53:58 localhost.localdomain systemd[1]: zoneminder.service: Failed with result 'timeout'.
Nov 02 11:53:58 localhost.localdomain systemd[1]: Failed to start ZoneMinder CCTV recording and security system.

```

knight-of-ni, 2018-11-02:
You are ahead of me. I have yet to upgrade to f29.

Based on your description, report the following:
- Provide the output of: `ps -o comm="" -p 1`
- ZoneMinder uses policy kit to allow non-root user apache to restart the service from the web console. Look for polkit errors: `sudo grep -ri zoneminder.*polkit /var/log/`
- Reboot. Try to start zoneminder from the command line with `systemctl start zoneminder`. Verify it is running properly with `systemctl status zoneminder`. Also verify the web console is working normally. Do *NOT* try to start/stop/restart from the Web Console for this step. Report your results.
- While zoneminder is running normally, now try to restart ZoneMinder from the Web Console. Report the results both from the web console and the command line with `systemctl status zoneminder`

knight-of-ni, 2018-11-02:
No, that isn't what you said initially. You said you couldn't start ZoneMinder *after your tried to restart the service from the web console*. What I am asking you do to verify the start/restart works from the command line *before* you try to manage the service from the web console.

I know this sounds like I am being picky, but this really matters. Please verify you did exactly this and in the order specified. We have seen these symptoms before.

Try widening your search for polkit messages with: `sudo grep -ri polkit /var/log/`. That might give a lot of results. If polkit is working, you should at least see something like this:
```
/var/log/secure-20181014:Oct 12 13:56:41 zoneminder polkitd[494]: Registered Authentication Agent for unix-process:13044:200585755 (system bus name :1.1264 [/usr/bin/pkttyagent --notify-fd 5 --fallback], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8)
```
Something else to check. When ZoneMinder refuses to start from the command line, check if there aren't already zm processes running with: `ps -A |grep zm`

Frankstar, 2018-11-02:
> No, that isn't what you said initially. You said you couldn't start ZoneMinder _after your tried to restart the service from the web console_. What I am asking you do to verify the start/restart works from the command line _before_ you try to manage the service from the web console.

well - you are not wrong - we speak about 2 differnet things.
Yes its true, the service was running - i stopped it via the webconsole and now he is refusing to start.
Doesnt matter if i try start from terminal or webconsole.
When i restart the server, i cant start it from terminal via `systemctl start zoneminder`
it produce the error in Description post. (not even touching the webconsole)

> I know this sounds like I am being picky, but this really matters. Please verify you did exactly this and in the order specified. We have seen these symptoms before.

its fine, just try to troubleshoot it with you.
im a sysadmin - so for me it was clear to check everything from terminal before i try something via the webconsole - sry.

> Try widening your search for polkit messages with: `sudo grep -ri polkit /var/log/`. That might give a lot of results. If polkit is working, you should at least see something like this:

not realy what we are looking for but i found this:
```
Nov  2 11:32:11 localhost pkexec[1401]: pam_unix(polkit-1:session): session opened for user root by (uid=48)
Nov  2 11:32:11 localhost pkexec[1401]: apache: Executing command [USER=root] [TTY=unknown] [CWD=/usr/share/zoneminder/www] [COMMAND=/usr/bin/perl /usr/bin/zmsystemctl.pl start]
```
also this might be the log when i stoped the server via webconsole:
```
Nov  2 10:33:27 localhost pkexec[2198]: pam_unix(polkit-1:session): session opened for user root by (uid=48)
Nov  2 10:33:27 localhost pkexec[2198]: apache: Executing command [USER=root] [TTY=unknown] [CWD=/usr/share/zoneminder/www] [COMMAND=/usr/bin/perl /usr/bin/zmsystemctl.pl stop]
```

but nothing else related to polkit and zoneminder (zm)

> Something else to check. When ZoneMinder refuses to start from the command line, check if there aren't already zm processes running with: `ps -A |grep zm`

sadly - nothing.

------ 
i cant reproduce why it started the second time tho

Frankstar, 2018-11-02:
`su -s /bin/bash -c '/usr/bin/zmpkg.pl start' apache` produce:

```
02.11.2018 14:34:38.199165 zmpkg[1698].INF [main:57] [Command: start]
02.11.2018 14:34:38.215887 zmpkg[1698].INF [main:305] [Sanity checking States table...]
02.11.2018 14:34:38.240214 zmpkg[1698].INF [main:97] [Command: start]
02.11.2018 14:34:38.884272 zmsystemctl[1703].INF [main:71] [Redirecting command through systemctl]
Job for zoneminder.service failed because a timeout was exceeded.
See "systemctl status zoneminder.service" and "journalctl -xe" for details.
```
`systemctl status zoneminder.service` = errors in description post.
so not realy more info.

knight-of-ni, 2018-11-02:
It does seem like polkit is working. I am upgrading my vm to f29 to try and duplicate the issue.

Frankstar, 2018-11-02:
its strange.
I just restarted the server and immediately started zoneminder -> it worked.
```
[root@localhost ~]# systemctl start zoneminder
[root@localhost ~]# systemctl status zoneminder
● zoneminder.service - ZoneMinder CCTV recording and security system
   Loaded: loaded (/usr/lib/systemd/system/zoneminder.service; disabled; vendor preset: disabled)
   Active: active (running) since Fri 2018-11-02 16:03:47 CET; 6s ago
  Process: 1156 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=0/SUCCESS)
 Main PID: 1166 (zmdc.pl)
    Tasks: 7 (limit: 4915)
   Memory: 138.8M
   CGroup: /system.slice/zoneminder.service
           ├─1166 /usr/bin/perl -wT /usr/bin/zmdc.pl startup
           ├─1194 /usr/bin/perl -wT /usr/bin/zmfilter.pl --filter_id=1 --daemon
           ├─1200 /usr/bin/perl -wT /usr/bin/zmfilter.pl --filter_id=2 --daemon
           ├─1204 /usr/bin/perl -wT /usr/bin/zmaudit.pl -c
           ├─1208 /usr/bin/perl -wT /usr/bin/zmwatch.pl
           ├─1217 /usr/bin/perl -w /usr/bin/zmtelemetry.pl
           └─1222 /usr/bin/perl -wT /usr/bin/zmstats.pl

Nov 02 16:03:40 localhost.localdomain systemd[1]: Starting ZoneMinder CCTV recording and security system...
Nov 02 16:03:47 localhost.localdomain systemd[1]: Started ZoneMinder CCTV recording and security system.
```
i restarded again - and immediately started zoneminder  again -> doesnt work ... got the 
`Unable to connect to server using socket at /var/lib/zoneminder/sock/zmdc.sock` error.

Frankstar, 2018-11-02:
found something.
dont know if its relevant.
but this gets logged everytime before i try start the service.
```
Nov  2 13:15:15 localhost systemd[1]: Starting ZoneMinder CCTV recording and security system...
Nov  2 13:15:18 localhost audit[3770]: ANOM_ABEND auid=4294967295 uid=48 gid=48 ses=4294967295 pid=3770 comm="zmdc.pl" exe="/usr/bin/perl" sig=11 res=1
Nov  2 13:15:18 localhost kernel: zmdc.pl[3770]: segfault at 91 ip 0000000000000091 sp 00007ffe4fab1858 error 14 in perl[55a6fc986000+1000]
Nov  2 13:15:18 localhost kernel: Code: Bad RIP value.
```

ileGITimo, 2018-11-03:
I'm having the same problem.  Commenting out the PID file does make it start but only one of my four camers is capturing.  Web console claims it stopped. Log is full of errors.
(Knight-of-NI from NI forums?)

ileGITimo, 2018-11-04:
With  PIDfile set systemd times out.  Without it systemctl returns, somehow it thinks zoneminde  is running,but it is not, as it hangs/dies after trying yo setup the first camera (zmc -m 1)., and  zmdc.pl core dumps.
`zmdc.pl[19979]: segfault at 91 ip 0000000000000091 sp 00007ffebb484688 error 14 in perl[56211b9c0000+1000]
kernel: Code: Bad RIP value.
Process 19979 (zmdc.pl) of user 48 dumped core.
Stack trace of thread 19979:
#0  0x0000000000000091 n/a (n/a)
`

alexanderhampel, 2018-11-06:
Same exact issue on my end after upgrading from F28 to F29...

ileGITimo, 2018-11-06:
There's a timing issue somewhere, where zmdc.pl exits after forking the first daemon, zmc in my case.  

I was able (after some minor code changes) to run almost everything through (several) perl -d sessions, and everything started fine.  

After undoing all my changes, starting the code via systemctl worked once, and then it never ran again.  This may explain why Frankstar seems to be able to run it some times.

As I was able to get a list of everything that started. I then manually started it, as apache, by running "zmdc.pl startup", followed by all the "zmdc start <daemon> <args>, to match what was running, and zm is running.

knight-of-ni, 2018-11-06:
@ileGITimo That's correct about the timing issue, and setting LOG_LEVEL_SYSLOG -> none changes the timing. Did you try that?

Frankstar, 2018-11-06:
where should we add the LOG_LEVEL_SYSLOG  = none ??

knight-of-ni, 2018-11-06:
> where should we add the LOG_LEVEL_SYSLOG = none ??

Click logging from the web console

m-theredhead, 2018-11-06:
I am having the same issue on Fedora 29 as well.    I tried setting the SYSLOG value to none but it still isn't able to start.

connortechnology, 2018-11-06:
It might not be syslog that is causing the problem.  Might be db logging.  I always turn my db logging to Warn.  Anyways, we will get this narrowed down eventually.

knight-of-ni, 2018-11-06:
When I set LOG_LEVEL_DATABASE to none, the problem goes away

ileGITimo, 2018-11-06:
Works here too, with not db logging.

m-theredhead, 2018-11-06:
Setting LOG_LEVEL_DATABASE to none allows zoneminder to start right away for me as well

alexanderhampel, 2018-11-07:
Same here. Everything works with logging set to None.  Thank you.

knight-of-ni, 2018-11-07:
If you would rather leave LOG_LEVEL_DATABASE enabled, you can ucomment this line in zmdc.pl: https://github.com/ZoneMinder/zoneminder/blob/master/scripts/zmdc.pl.in#L441

This will get you back on track with no loss of functionality. Note that this is still not the real fix. 
A real fix is being developed and it looks like it is going to be more  significant than we initially thought.

closed 2021-02-03 by connortechnology