ZoneMinder/zoneminder #2294: %EI1% and %EIM% email return error 69

aaronsta1, 2018-11-07:
i see this was an issue before and its marked as fixed but i am still getting an issue where the pictures are not attached to the email.

the log says something like this

2018-11-07 01:45:51 | zmfilter_3 |   | 13707 | WAR | No image for EIM | zmfilter.pl |  
2018-11-07 01:45:51 | zmfilter_3 |   | 13707 | ERR | Failed ffmpeg -ss 161.68 -i '/mnt/zm0/1/2018-11-07/5308/5308-video.mp4' -frames:v 1 '/mnt/zm0/1/2018-11-07/5308/01104-capture.jpg' status 69 | zmfilter.pl

aaronsta1, 2018-11-07:
i think i have worked out the issue but i am unsure of how to fix or address it.
i am using the mocord setting and i think im getting this error because the mp4 file is still in use? 
ive lowered the recording time down from 600 seconds to 300 seconds and now it mostly only happens if the alarm is at the beginning of the video.

TheSwert, 2018-12-12:
I'm experiencing the same issue. It started happening when I upgraded from Ubuntu 16.04 to 18.04. I confirmed it was working on version 1.32.3 prior to the upgrade. After upgrading the distro, I had to run the below commands to make Zonemider function again.

```
awk '$0="date.timezone = "$0' /etc/timezone >> /etc/php/7.2/apache2/php.ini
a2enmod php7.2
```
I'm wondering if something else needs to be done to fix the functionality. Besides the failure to send emails, everything else appears to be working.

I'm getting errors in the log similar to below

Failed ffmpeg -ss 7.48 -i '/var/cache/zoneminder/events/2/2018-12-12/4198465/4198465-video.mp4' -frames:v 1 '/var/cache/zoneminder/events/2/2018-12-12/4198465/00076-capture.jpg' status 69

adamcoffee1, 2019-03-26:
Just wanted to add my experience with this error.  I've got several 1080p IP cameras passthrough mocording to an NFS share.  I've played with a few things, just trying to see if any behavior changes:

- Set NFS mount in /etc/fstab to sync ([default is async](https://linux.die.net/man/5/nfs))
- Decrease event length from 600 seconds to 300 seconds
- Increase filter execution interval to 600 seconds

(The last two are an attempt to get the filter to only grab the frame well after the event has finished and recording has moved on to a new one.)

It's an interesting problem as it appears that the MP4 file hasn't been committed to storage prior to FFMPEG being executed by the filter.  If I take the failed command and run it myself _right after it failed to run_, it completes successfully.

Regardless of this, I'm absolutely loving ZM 1.32.3; much improvement and you really cannot find this degree of freedom in configuration with other CCTV solutions. :)

connortechnology, 2019-03-26:
I think I have an idea what is up.  in 1.32. In 1.32 I changed the filter query so that it could operate on events that are still being recorded. So now I think you need to add another filter term that checks that the end date/time is not NULL

adamcoffee1, 2019-03-26:
Oooh that sounds very promising; I am going to try this now and will report back!  Thanks

adamcoffee1, 2019-03-26:
OK, so I think it may be something more obscure.  Here is debug logs from zmfilter, as it finds an event to be attached and emailed:

```
03/26/2019 17:33:09.206596 zmfilter_5[20135].INF [main:286] [Checking filter EmailMotion email returned 0 events ]
03/26/2019 17:33:09.217881 zmfilter_5[20135].DBG [main:215] [Sleeping for 30 seconds]
03/26/2019 17:33:39.220463 zmfilter_5[20135].DBG [ZoneMinder::Filter:114] [Filter::Execute SQL (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND (   M.Id in (2,3,4)   and   E.AlarmFrames > 2   and   E.Cause != 'Signal'   and   E.EndTime IS NOT NULL   ) and ( E.Emailed = 0 ) order by E.Id desc limit 0,10)]
03/26/2019 17:33:39.321963 zmfilter_5[20135].DBG [ZoneMinder::Filter:128] [Loaded 1 events for filter EmailMotion using query (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND (   M.Id in (2,3,4)   and   E.AlarmFrames > 2   and   E.Cause != 'Signal'   and   E.EndTime IS NOT NULL   ) and ( E.Emailed = 0 ) order by E.Id desc limit 0,10)]
03/26/2019 17:33:39.322669 zmfilter_5[20135].INF [main:286] [Checking filter EmailMotion email returned 1 events ]
03/26/2019 17:33:39.337527 zmfilter_5[20135].DBG [main:296] [Checking event 51251]
03/26/2019 17:33:39.338861 zmfilter_5[20135].INF [main:766] [Creating notification email]
03/26/2019 17:33:39.355519 zmfilter_5[20135].DBG [main:649] [Frames: rows: 19 first alarm frame: HASH(0x55af519b0200) max_alaarm_frame: HASH(0x55af519b0200), score: 255]
03/26/2019 17:33:39.356280 zmfilter_5[20135].DBG [ZoneMinder::Object:114] [Loading ZoneMinder::Storage from Storage WHERE Id = 2]
03/26/2019 17:33:40.500715 zmfilter_5[20135].DBG [main:457] [Output: ]
03/26/2019 17:33:40.501773 zmfilter_5[20135].ERR [main:460] [Failed ffmpeg -ss 2.00 -i '/mnt/zm/2/2019-03-26/51251/51251-video.mp4' -frames:v 1 '/mnt/zm/2/2019-03-26/51251/00019-capture.jpg' status 69]
03/26/2019 17:33:40.514423 zmfilter_5[20135].INF [main:774] [Sending notification email 'ZoneMinder: Alarm - Parking Space-51251']
03/26/2019 17:33:40.997942 zmfilter_5[20135].INF [main:844] [Notification email sent]
03/26/2019 17:33:41.100791 zmfilter_5[20135].DBG [main:215] [Sleeping for 30 seconds]
```
First thing to note, it appears a condition is already automatically added to filter out events with a NULL ending date/time.  You can see it is the first condition in the WHERE, and then the last condition is what I added to test:

`WHERE (E.EndTime IS NOT NULL) AND (   M.Id in (2,3,4)   and   E.AlarmFrames > 2   and   E.Cause != 'Signal'   and   E.EndTime IS NOT NULL   ) and ( E.Emailed = 0 )`

At 17:33:40.501773 ffmpeg is called and fails.  I manually copied that command and ran it in my terminal a few seconds later, and it succeeded.  Now that I think about it, if the file simply didn't exist, we'd get a typical "file not found" error from ffmpeg.  Instead it spits back 69...

I could not figure out how to understand ffmpeg error codes from their source, so I manually edited zmfilter to call ffmpeg while redirecting stdout and stderr to a location it could write and I could read.  Here is a log of ffmpeg *failing* as we have been seeing:

```
ffmpeg version 3.4.4-0ubuntu0.18.04.1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/mnt/zm/2/2019-03-26/51344/51344-video.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    title           : Zoneminder Security Recording
[... cut ...]
At first this made me think the video coming from my camera (passthrough) is somehow corrupted or not to spec.  But in both cases, the resulting event videos can be viewed without issues in the web console or in VLC after downloading.  (Plus, as noted before, the command will succeed when ran manually a few seconds later.)  No glitches or anything goofy.

The only idea I have left is that there is some race where ffmpeg is reading the file before it has been completely written.  Is that even possible?  Other than that, perhaps it's something to do with us using an NFS share?  I'm out of ideas but hopefully this can be of help.

connortechnology, 2019-03-27:
It is possible.  Currently zma ends the event, updates the db and zmc has to notice that the event has closed and then finishes writing out the file.  

Maybe add another condition that only reports events that are more than a minute old.  So EndTime < -1 minute

adamcoffee1, 2019-03-28:
OK so I edited my motion email filter to have an extra condition that End Date/Time < -5 min, in the hopes of giving zmc plenty of time to finish up with writing prior to zmfilter trying to read.  Note that for testing purposes, I've also reduced the event length of the camera I am using for testing to 200 seconds.  (Ought to be plenty of time delta between event finish and filter trigger?)  Here's a log of zmfilter picking up an event with the new filter:

```
03/28/2019 11:44:36.010149 zmfilter_5[6916].DBG [main:215] [Sleeping for 30 seconds]
03/28/2019 11:45:06.015013 zmfilter_5[6916].DBG [ZoneMinder::Filter:114] [Filter::Execute SQL (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND (   M.Id in (2,3,4)   and   E.AlarmFrames > 2   and   E.Cause != 'Signal'   and   E.EndTime < '2019-03-28 11:40:06'   ) and ( E.Emailed = 0 ) order by E.Id desc limit 0,10)]
03/28/2019 11:45:06.121274 zmfilter_5[6916].DBG [ZoneMinder::Filter:128] [Loaded 1 events for filter EmailMotion using query (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND (   M.Id in (2,3,4)   and   E.AlarmFrames > 2   and   E.Cause != 'Signal'   and   E.EndTime < '2019-03-28 11:40:06'   ) and ( E.Emailed = 0 ) order by E.Id desc limit 0,10)]
03/28/2019 11:45:06.121831 zmfilter_5[6916].INF [main:286] [Checking filter EmailMotion email returned 1 events ]
03/28/2019 11:45:06.135114 zmfilter_5[6916].DBG [main:296] [Checking event 53384]
03/28/2019 11:45:06.136039 zmfilter_5[6916].INF [main:766] [Creating notification email]
03/28/2019 11:45:06.153160 zmfilter_5[6916].DBG [main:649] [Frames: rows: 55 first alarm frame: HASH(0x55c05fb849a0) max_alaarm_frame: HASH(0x55c05fb849a0), score: 255]
03/28/2019 11:45:06.153526 zmfilter_5[6916].DBG [ZoneMinder::Object:114] [Loading ZoneMinder::Storage from Storage WHERE Id = 2]
03/28/2019 11:45:07.252625 zmfilter_5[6916].DBG [main:457] [Output: ]
03/28/2019 11:45:07.253714 zmfilter_5[6916].ERR [main:460] [Failed ffmpeg -ss 5.20 -i '/mnt/zm/2/2019-03-28/53384/53384-video.mp4' -frames:v 1 '/mnt/zm/2/2019-03-28/53384/00053-capture.jpg' status 69]
03/28/2019 11:45:07.278145 zmfilter_5[6916].INF [main:774] [Sending notification email 'ZoneMinder: Alarm - Parking Space-53384']
03/28/2019 11:45:07.850027 zmfilter_5[6916].INF [main:844] [Notification email sent]
03/28/2019 11:45:07.881068 zmfilter_5[6916].DBG [main:215] [Sleeping for 30 seconds]
```
Seems it's still failed.  Interesting as some work, and some don't.  Argh!

Now here's another thought.  Since the `-ss` option of ffmpeg is [seeking](https://trac.ffmpeg.org/wiki/Seeking) using keyframes, maybe the frequency of keyframes in my camera's stream is causing this?

The camera I'm testing with is a cheap, but surprisingly decent for the price SV3C [SV-B11VPOE-1080P-L](http://www.sv3c.com/1080P-POE-Varifocal-Bullet-IP-Camera-SV-B11VPOE-1080P-.html).  Here's the available encoder settings:
[image]

Originally, the "Frame Interval" as they describe it (I'd call it a Key Frame or I Frame) was set to 40.  Meaning every 40 frames, a key frame comes down the pipe.  At 10FPS, that is one key frame every 4 seconds.  That sounds too infrequent to me.

The range given for selection is 1~200 frames.  (Kind of funny you can just tell it to make every frame a key frame eh?)  I set it to 10, so one key frame every second.  Watching the live stream, you can see the slight "twitch" in pixelation when the key frame comes, so it's definitely working as expected.

And what do you know - every email has _succeeded_ in attaching since then, with no extra EndTime condition!  So my overall theory is that ffmpeg is basically giving up on finding a keyframe if the `-ss` parameter given happens to fall far away from a key frame.  When I had a key frame coming every 4 seconds, it would scroll through (hence the thousands of lines of "NAL" errors) and give up.

At the end of the day I do not think there is anything ZM is doing wrong.  It's simply a characteristic of how ffmpeg seeks and H264 operates.  This is exacerbated by our well-meaning tendency to turn down FPS settings on our cams.  (Who needs 30FPS when 10FPS or even 5 will do?)  What we're forgetting is to _also_ adjust down the key frame interval to ensure the stream is being "refreshed" often enough.  I would assume this increases file size somewhat, but oh well, H264 is still a massive improvement over storing JPG's.  I have a hunch this also improves the video quality of any motion in a given frame - when you spend too long without a key frame, the "[difference frames](https://en.wikipedia.org/wiki/Video_compression_picture_types#Bi-directional_predicted_(B)_frames/slices_(macroblocks))" are sort of "stale":
[image]

Now what to do about it?  Since this is really up to the user to set their cameras up as they see fit, I'm thinking I could add a caveat section [like this](https://zoneminder.readthedocs.io/en/stable/userguide/filterevents.html#caveat-with-relative-items) to the docs that describes why people may encounter failures to attach images.  I'd like to test some more with my other cams first, and of course hear your opinion on whether this is the best solution!

adamcoffee1, 2019-03-28:
One issue I've noticed with this is that some video sources don't have a key frame option.  For example my doorbell cam just doesn't have it:
[image]

Maybe I can try X264 encoding (basically re-encoding with more frequent key frames).  Because when encoding sources within ZM, it looks like I can use the `--keyint` option to specify how often a key frame happens.

For example, I have a couple analog cams that are digitalized on one ZM server, and then streamed to my main server for storage.  The stream comes in as jpeg, so I'm using the X264 Encode to store it as video.  Since the video is only ~3FPS, I'm using the following with good success, to get a keyframe about once per second, and working email attachments:
[image]

Finally, it may be worth exploring the difference between "input" and "output" [seeking](https://trac.ffmpeg.org/wiki/Seeking) with ffmpeg.  ZM currently does input seeking, but from the how the article reads, output seeking _may_ correct these issues, at the cost of greater processing effort as it moves frame by frame, not just jumping by key frames.  To do output seeking, you place the `-ss` parameter _after_ the `-i` parameter, not before it.

Guess I'll test that next with an edit to zmfilter!

bieber, 2019-05-19:
Just to add a data point, I'm also currently experiencing this on v1.32.3.  I'm also using SV3C cameras, and I set the frame rate to 10fps and the keyframe interval to 10 like @adamcoffee1, but unfortunately I'm not seeing the same success.  Even with my filter set to run with end time at -5 minutes, most of my event emails fail to generate an image.  Some do, but it seems like a crapshoot which ones will or won't

adamcoffee1, 2019-05-21:
Interesting - if you don't mind what camera models do you have? And are you also using CBR with High Profile? Since making my changes I have yet to see a single motion email sourced from my SV3C's that failed to include an image.

bieber, 2019-05-21:
I am on high profile, but VBR encoding on the Good preset.  Cameras are SV-B06POE-1080P-A

adamcoffee1, 2019-05-25:
The only thing I can think of is that the choice of VBR versus CBR may be playing into this. Of course it's not optimal for network/storage utilization, but perhaps try out CBR and see if that makes the difference?

bieber, 2019-05-25:
Yeah, I'll give it a shot.  Getting more timely alerts is more valuable than retention time to me anyways

bieber, 2019-05-26:
Weird, still no luck.  I think I'm just gonna write my own mailer script, pull frames out of the video at fixed intervals and attach them to the email

thunderbolt-tom, 2019-06-07:
TL;DR
I've done lots of testing on this issue, and it's nothing to do with corrupt files or ffmpeg being run too soon after an event finishes. 

My testing shows that ffmpeg fails for both video generation (OPT_FFMPEG) and image generation (the %EIM% tag etc) only when ffmpeg is run from within a filter that is being run automatically due to the "Run filter in background" option being selected.

If the "Run filter in background" option is deselected and the filter is run manually then ffmpeg will succeed. This is not a solution, but hopefully points towards what is causing the problem. 

-----------------------------

I also have the same issue after recently upgrading to v1.32.3 and starting to use video storage. I've been trying to figure out what's going on.

Firstly I get the same "No image for EIM" issue when trying to send emails _almost_ all of the time, although very occasionally it does work. I also have OPT_FFMPEG switched on and I've activated the "Create videos for all matches" in the "email" filter to create smaller videos to attach to the email. This also fails every time with the ffmpeg log showing the same sorts of error messages that other users have seen, so I suspect it's the same cause:

```
[h264 @ 0x562f7b42a340] Invalid NAL unit size (2088513 > 8158).
[h264 @ 0x562f7b42a340] Error splitting the input into NAL units.
```

The one slight difference with the video creation is that it does actually create an output file, but it only contains 1 frame, and ffmpeg still exits with the "Conversion failed!" error.

As other users have mentioned, if I try and run the same ffmpeg command that Zomeminder is running then it works fine. I've also tried running ffmpeg under the www-data user (which is what Zoneminder runs as on Ubuntu) and it also works fine, so both of these, for example, work:

```
/usr/bin/ffmpeg -y -r 10.17  -i 168082-video.mp4 -s  1280x720 -c:v libx264 -preset fast -crf 38 -vf hqdn3d=8:6:12 ~/zmtest.mp4
sudo -u www-data /usr/bin/ffmpeg -y -r 10.17  -i 168082-video.mp4 -s  1280x720 -c:v libx264 -preset fast -crf 38 -vf hqdn3d=8:6:12 zmtestwww.mp4
```

I've also tried running the zmvideo.pl and zmfilter.pl scripts from the command line to instigate video creation, and these too work. So for example running all of the following from the command line result in the video successfully being created:

```
zmvideo.pl -e 168082 --format mp4
zmvideo.pl --filter_name=email --format mp4
zmfilter.pl --filter=email
```

So in summary it seems that any method of running ffmpeg on the command line works, even using the Zoneminder scripts. this implies that the event video file itself is not corrupt in any way, that Zoneminder is generating the correct parameters for ffmpeg, and that ffmpeg itself is not causing the problem.

However whenever the video or image should be created during the email filter run, it always fails. I've tested various scenarios to see what makes a difference. One comment was that it might be due to ffmpeg being run too soon after the event ends and the file might not be closed properly. Adding a 5 minute delay to end date/time has already been shown to make no difference. I tried further by choosing an event that was many hours old and setting the Videoed field to 0 in the Events table which causes Zoneminder to try and create the video again on the next filter run. ffmpeg still fails, so I'm fairly confident that the issue isn't because ffmpeg is being run too soon.

Another scenario I tested was to transcode the event video on the command line and force Zonminder to run ffmpeg as above to again check that it's not caused by some kind of corruption:

```
sudo -u www-data mv 168224-video.mp4 168224-video.mp4.orig
sudo -u www-data ffmpeg -i 168224-video.mp4.orig -c:v libx264 168224-video.mp4
```

Although the above command worked, ffmpeg still failed when Zonminder ran the filter. However, as the ffmpeg error relates to decoding the h264 stream I tried transcoding to mjpeg:

`sudo -u www-data ffmpeg -i 168224-video.mp4.orig -c:v mjpeg 168224-video.mp4`

This time ffmpeg succeeds when run by the Zoneminder filter! Unfortunately it's not a solution as Zoneminder's events are stored in h264, not mjpeg, but it does narrow down the problem to the x264 decoding in ffmpeg.

In another test I created a very simple bash script to pass through the ffmpeg command. The script is as follows:

~/scripts/ffmpeg_passthrough.sh
```
#!/bin/bash
echo "Command: $*"
ffmpeg "$@"
```

It just prints out the options that are passed to this script, then runs ffmpeg itself with those options. I made this executable by everyone and put the script path in PATH_FFMPEG. If I run it from the command line as so then it works:

`~/scripts/ffmpeg_passthrough.sh -y -r 10.17  -i 168082-video.mp4 -s  1280x720 -c:v libx264 -preset fast -crf 38 -vf hqdn3d=8:6:12 ~/zmtest.mp4`

But again when Zoneminder runs the filter, it fails with the same errors. What I can now do is see what happens if I put an ffmpeg command that acts on a random video file unrelated to Zonminder so I edited the script as so:

```
#!/bin/bash
echo "Command: $*"
ffmpeg -y -i /home/user/unrelated_video.mp4 -c:v libx264 -preset veryfast -t 30 /home/user/unrelated_output.mp4
```

When Zonminder runs the filter then even this command fails. This proves that the error is completely unrelated to the event video file. Interestingly this file contained aac audio, and the ffmpeg log also showed aac decoding errors in addition to the h26 [... cut ...]
So it appears to be a problem to do with decoding h264 caused when Zoneminder is trying to run ffmpeg (either for video or image generation) but only from within a filter that is run in the background. I can't begin to fathom why only the decoding part of ffmpeg is affected (I'd expect ffmpeg to either run and work or to fail to run at all).

I also don't know much about the inner workings of Zoneminder so I don't know how a filter with "Run filter in background" is executed differently to a filter that is manually executed, but hopefully my testing can help someone who is more familiar with Zoneminder to find a fix.

thunderbolt-tom, 2019-06-08:
I've done some more thorough testing and I think I've found the cause of the issue and a potential solution, although I'm quite hazy on the details.

I've focussed on trying to replicate the errors from ffmpeg by looking at how Zoneminder runs ffmpeg when "Run filter in background" is selected. Zoneminder as a whole is started from zmpkg.pl which also starts zmfilter.pl running as a daemon for each filter that runs in the background. However it doesn't run zmfilter directly, it starts it through zmdc.pl. zmdc is quite complex, but one thing I noticed was that it closes file descriptors before starting a daemon process, and it's this that causes the issue.

To demonstrate, the following simple Perl script will cause the exact same ffmpeg errors. It's the closing of stdin that causes the problem, stdout and stderr are fine to close.

```
#!/usr/bin/perl -w

use v5.10;
use POSIX;

my $command = "ffmpeg -y -i 168224-video.mp4 -c:v libx264 -preset veryfast 168224-video-out.mp4 > ffmpeg.log 2>&1";

POSIX::close(0); # close stdin, if commented out then it works

exec($command);

```

It's a bit more complicated than this for two reasons: firstly there is already a potential fix in zmdc.pl https://github.com/ZoneMinder/zoneminder/commit/66f11435b990f5399066fc70ed353b992a2998e1 which specifically doesn't close stdin, stdout and stderr, but clearly it doesn't solve the problem in this case. Secondly, Zoneminder will actually run the ffmpeg command using sudo (in order to change the user), and as far as I can tell using sudo should create a new shell with its own file descriptors. To demonstrate, if you add sudo before ffmpeg in the command in the script above then it will work even though stdin is being closed, yet within Zoneminder it still doesn't work. So clearly there is something far more complex than I understand going on.

However, after a bit of searching I found http://mywiki.wooledge.org/BashFAQ/089. it doesn't describe exactly what we're seeing, but it does states that for reasons unknown ffmpeg is trying to read from stdin, and to prevent ffmpeg doing so you should add `< /dev/null` to the end of the ffmpeg command.

Reading further there's actually an ffmpeg FAQ question about this http://ffmpeg.org/faq.html#How-do-I-run-ffmpeg-as-a-background-task_003f and ffmpeg has the option `-nostdin` which essentially does the same as `< /dev/null`. Adding it to the script above worked, so I went ahead and modified the ffmpeg commands in Zonminder (/usr/share/perl5/ZoneMinder/Event.pm line 350 for video generation and /usr/bin/zmfilter.pl line 451 for image generation) and success, videos and images are now created without issue.

bieber, 2019-06-10:
Brilliant @thunderbolt-tom, adding the -nostdin flag did the trick.  I'm getting images in all my emails now.  Is anyone from the project in this thread?  Would y'all accept adding that flag to the commands as an acceptable fix?

connortechnology, 2019-06-10:
yeah that seems cool.

connortechnology, 2019-06-10:
Our scripts should not depend on stdin/out etc.

commit 2e624522 (2019-06-10): fixes #2294 (#2637)
closed 2019-06-10 by connortechnology