ZoneMinder/zoneminder #2316: Streaming broken behind reverse proxy

briantroy, 2018-11-19:
https://github.com/ZoneMinder/zoneminder/blob/71bfd50b2c92c26846026b581fdf791acda1fa57/web/includes/Monitor.php#L295

To make streaming work with a reverse proxy I modified this line to be:
      if ( isset($_SERVER['HTTP_X_FORWARDED_SERVER']) ) {
         $streamSrc .= $_SERVER['HTTP_X_FORWARDED_SERVER'];
      } else {
         $streamSrc .= $_SERVER['SERVER_NAME'];

Other changes may be required in the code directly above.

abishai, 2018-11-21:
I doubt any code changes required: reverse proxy should pass Host header.

briantroy, 2018-11-21:
It was not working for me. All other URIs had the correct host name but not in the case of the monitors. The IP of the proxied host was in the URI causing all monitors to fail. 

Here is my proxy config:

ProxyRequests Off
ProxyVia On

<Proxy *>
   Require all granted
</Proxy>

ProxyPass /zm http://192.168.86.18/zm
ProxyPassReverse /zm http://192.168.86.18/zm

briantroy, 2018-11-21:
I'd also note that this proxy fronts 6 other http services and all the others work correctly.

cotman, 2018-12-01:
Has (and still does) work for me with the following:

    location /zm/ {
        include /etc/nginx/proxy_params;
	proxy_set_header X-Forwarded-Proto https;
	proxy_pass https://192.168.1.15/zm/;
        proxy_redirect https://192.168.1.15/zm/ https://host/;
}

...I wouldn't claim to know the ins an outs of reverse proxying with nginx, but the above does work.

closed 2018-12-16 by briantroy

faginbagin, 2019-02-28:
briantroy, since you closed this issue, I assume you found a solution. It sure would be nice to know what it was. I'm having the same problem.

briantroy, 2019-03-02:
faginbagin - here is my working apache proxy configuration. This works and doesn't require any changes to the Zoneminder code:

ProxyRequests Off
ProxyVia On
<Proxy *>
   Require all granted
</Proxy>

ProxyPreserveHost On
ProxyPass /zm http://xxx.xxx.xxx.xxx/zm
ProxyPassReverse /zm http://xxx.xxx.xxx.xxx/zm

faginbagin, 2019-03-04:
briantroy: Many thanks for replying, but I still have a problem and some questions for you.

The config you provided is identical to the config you provided in comment https://github.com/ZoneMinder/zoneminder/issues/2316#issuecomment-440795551
including what I believe is a syntax error, since both are missing a closing `</Proxy>` tag.
Am I correct in assuming that tag belongs right after the Require directive?

I think I just figured this one out. Your post probably contained the closing tag, but had no markup, so it was dropped from the post. My closing tag was dropped from the preview until I surrounded it with back ticks. I would still like to confirm you've got it just after the Require directive.

I have modified my config file: /etc/apache2/conf-available/zoneminder.conf (symlinked to conf-enabled/zoneminder.conf using a2enconf) to this:
```
ProxyRequests Off
ProxyVia On
<Proxy *>
Require all granted
</Proxy>
ProxyPreserveHost On
ProxyPass /zm http://192.168.1.162/zm
ProxyPassReverse /zm http://192.168.1.162/zm
```
It hasn't helped. So maybe there's some other subtle difference For example, is your zoneminder configuration in a conf file, outside of any other container, like VirtualHost?

You posted your config on November 21 and didn't close the issue until December 16. Were you having the problem on Nov 21, with that config? Or was it working at that time? If it wasn't working on Nov 21, what changed between then and Dec 16?

Another point of comparison. My internal zoneminder server on 192.168.1.162, is running ubuntu 18.04 with apache 2.4, mariadb, phpmyadmin, and zoneminder 1.32.3 installed via Issac Connor's PPA:
https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-1.32
My last dist-upgrade was on March 1. How does that compare to your zoneminder server?

My external web server is running on Debian Stretch with apache 2.4.25 and OpenSSL 1.0.2q, with a letsencrypt wildcard certificate. It supports other apps like web mail, a photo gallery, and access to email over IMAPS. That's probably different from your environment, but I doubt it explains why streaming over https is working for you and not for me. Could I be wrong?

briantroy, 2019-03-04:
faginbagin

You are correct. The cut and paste dropped the closing </Proxy>.

The key change is “ProxyPreserveHost On”

I am in fact suing VirtualHosts so the file resides in /etc/apache2/sites-enabled (or is symlinked there to be 100% accurate). 

Nothing else in your setup is an obvious concern.

faginbagin, 2019-03-05:
It just dawned on me that I've been reading more into this issue than was there. Streaming with reverse proxy IS working for me, but only if the protocol is HTTP on both external and internal server. What isn't working is when the protocol is HTTPS to external server and HTTP to internal server. I now realize you never mentioned HTTPS. Sorry for the noise. Guess it's time for me to post on the forums.

raomin, 2019-04-29:
Hi. I have the same problem and I believe this issue is not noise.
I reverse proxy access to zoneminder with nginx in frontend in https.

Here is the nginx conf part for zm:

```
    location /zm {
        proxy_pass http://localhost:80/zm;
        proxy_set_header Host $host;
#        proxy_redirect http:// https://;
        proxy_http_version 1.1;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection $connection_upgrade;
        proxy_set_header        X-Forwarded-Proto https;
        proxy_set_header        SSL_PROTOCOL $ssl_protocol;
    }
```

Most of the UI works, but the stream doesn't. It appears that the url for some element is in HTTP and not in HTTPS as it should. Stream works fine when the whole site is accessed in http.
The web console shows mixed_content error:

`Mixed Content: The page at 'https://*****/zm/index.php?view=watch&mid=1' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://*****/zm/index.php?view=request&request=stream&connkey=18199&command=99'. This request has been blocked; the content must be served over HTTPS.
`

Somewhere in ZM code, there is probably a hardcoded http:// link, when it should rely on the added X-Forwarded-Proto.

raomin, 2019-04-29:
Actually, request `http://*****/zm/index.php?view=request&request=stream&connkey=18199&command=99` does not seem to be an issue for the live feed.
It is the url of the `img` html element that point to the http url:
`<div id="imageFeed">[image]</div>`

I would expect it to be defined by the `X-Forwarded-` headers.
Should I specify a base url manually somewhere?

faginbagin, 2019-04-29:
I agree that the inability to use a reverse proxy where you want https between the external server and the outside world and http between the external server and the internal zoneminder server is an issue. It's just not the issue that prompted briantroy, the OP, to create this issue. That's why I apologised for the noise.

FWIW, I've given up on this configuration. I've decided to take the path of least resistance and am running zoneminder on the external server because it works, including the ability to stream live video over https. I suspect part of the problem for zoneminder and reverse proxies with https is the need for zoneminder to support multiple servers. I wonder if https and live streams are an issue for multi-server configurations, but no more than wonder. I'm not using multiple servers and don't expect to.

abishai, 2019-04-29:
I run ZM with nginx making TLS termination and nginx as reverse proxy. Yes, ZM has some rough edges, but nothing that can't be solved without configuring.

raomin, 2019-04-29:
@abishai good to know you manage to reverse proxy ZM with https and get streaming working...!
Care for sharing your configuration to make it work?

abishai, 2019-04-29:
I think, it just works. Some non-important thing like fps request query is broken, but streaming itself works.
```
server {
    listen 443 ssl http2;
    server_name xxxxxxxxxxxxxx;

    ssl on;
    ssl_certificate xxxxxxxxxxxx;
    ssl_certificate_key xxxxxxxxxxxxxx;

    add_header Strict-Transport-Security "max-age=15768000; includeSubDomains; preload;";
    add_header X-Frame-Options "SAMEORIGIN";
    add_header X-Content-Type-Options "nosniff";
    add_header X-Permitted-Cross-Domain-Policies none;
    add_header X-XSS-Protection "1; mode=block";

    location / {
        proxy_send_timeout   180;
        proxy_read_timeout   180;
        proxy_pass http://10.0.4.6;
        proxy_next_upstream error timeout invalid_header http_500 http_502 http_503 http_504;
        proxy_redirect off;
        proxy_set_header        Host            $host;
        proxy_set_header        X-Real-IP       $remote_addr;
        proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
        }
}
```

```
server {
    listen 80;

    root /usr/local/www/zoneminder;
    index index.php;
    gzip off;

    location / {
        return 301 /zm;
    }

    location /cgi-bin/nph-zms {
            return 301 /zm/cgi-bin/nph-zms?$args;
    }

    location /zm/cache {
        alias /var/cache/zoneminder;
    }

    location /zm {
        alias /usr/local/www/zoneminder;

        location ~ /zm/.+\.php$ {

            try_files $uri =404;

            fastcgi_split_path_info ^(.+?\.php)(/.*)$;
            fastcgi_param SCRIPT_FILENAME $request_filename;
            fastcgi_index index.php;
            fastcgi_pass unix:/var/run/php-fpm.sock;
            include fastcgi_params;
        }

        location ~ \.(jpg|jpeg|gif|png|ico)$ {
            access_log  off;
            expires     33d;
        }

        location /zm/api/ {
            rewrite ^/zm/api(.+)$ /zm/api/app/webroot/index.php?p=$1 last;
        }

        location /zm/cgi-bin/nph-zms {
                include fastcgi_params;
                fastcgi_param SCRIPT_FILENAME $request_filename;
                fastcgi_pass  unix:/var/run/fcgiwrap/fcgiwrap.sock;
        }
    }
}
```

Barborica-Alexandru, 2020-01-03:
While I see some people may have some sort of workaround for this issue the truth is simply zoneminder does not support TLS offloading. I use Envoy and this issue persists. The only true solution is to simply not use TLS offloading and simply generate a self signed certificate on the webserver serving zoneminder. But this should be marked with a won't-fix as it is most definitely not closed.