ZoneMinder/zoneminder #2317: events are out of order.

aaronsta1, 2018-11-20:
my events are happening out of order and sometimes shown on different cameras.

say i have event 105 on camera one, if i do a gapless events playback and view the alarm, it will switch to camera two on  event 102 after 105 has completed playing. if i keep going it might go to 103 on camera 2 104 on camera one BEFORE the alarm happened, then back to 106.. 

i never had this before in previous versions.

connortechnology, 2018-11-20:
We need to know how you are getting to the events list.  When you view a list of events, it basically makes a filter and passes that around in the URL.  It sounds like your list is losing the monitor specification.  

So we need to know where you are clicking to get to the initial event view.  Then perhaps the contents of the url bar which will give us the filter that it is using.

aaronsta1, 2018-11-21:
just click on the monitor feed then click on past events.. then when you play back on gapless events it doesnt always choose the next event from that camera when it gets to the end.

commit 5b211d25 (2018-11-23): Merge pull request #2322 from connortechnology/fix_2317

fix #2317
commit 6128d2a4 (2018-11-23): fix #2317 by updating the url encoding of a filter to match events for that monitor
closed 2018-11-23 by knight-of-ni