ZoneMinder/zoneminder #2326: Montage Review Timezone

nitrag, 2018-11-28:
When using Montage Review History mode, if you are in a different timezone than your Zm instance, you can never catch-up to the last few hours. For PST to EST I can only see footage up until my local time (PST) which is 3 hours behind (EST) therefore I'm missing 3 hours of (the latest) footage.

v1.32.1

pliablepixels, 2018-12-06:
@mfauvain 
This may or may not be related to the OP issue. If you are sure its not a misconfiguration on your system, you should post a new issue. I'll post my thoughts there. Please also add your PHP version.

pliablepixels, 2018-12-06:
Please post a new issue - you posted in an issue that is not related to your problem and I don't want to discuss unrelated items in this issue.

closed 2019-02-14 by stale[bot]