ZoneMinder/zoneminder #2327: Problem with login.json and authentication relay

pliablepixels, 2018-11-29:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.2
- Built from source `do_debian.sh`
- Ubuntu 18

**Describe the bug**

Users who switched to the new version of zmNinja that uses `login.json` are complaining streaming breaks. In all cases, it is because they have AUTH_RELAY set to "none". One user told me today it was recommended they set it to "none" to fix some console error (not sure about this)

There is a problem with how login.json works today. The summary is if you use `OPT_AUTH` and `AUTH_HASH_LOGINS` but set `AUTH_RELAY` to `none`, the credentials return `user=username`. This is wrong because streaming breaks. 

Suggested solutions:

- We append password to the `authRelay=none` situation too [here](https://github.com/ZoneMinder/zoneminder/blob/master/web/api/app/Controller/HostController.php#L78)

- However, I am not sure why we are using authRelay as a check at all. Should we not be simply using `AUTH_HASH_LOGINS` instead and if so, return `&auth=` irrespective of what auth Relay is set at?

knight-of-ni, 2018-11-29:
When streaming only works when AUTH_RELAY is set to none, that means the end user has a time related issue with their configuration. Setting this to none is a workaround that does not fix the underlying problem. While one can find threads in the user forum where someone set this value to none and reported "success", this was never a recommended solution by us.  The recommended solution has always been to fix the underlying time issue. They should fix that problem first before doing anything else.

connortechnology, 2018-11-29:
I think we should remove all these options and just always use auth_hash.  There really is no reason not to.

pliablepixels, 2018-11-29:
Ok thanks. There are two points in this issue

1. Key point: If `AUTH_HASH_LOGIN` is on, and `AUTH_RELAY` is set to `None`, what should `login.json` return in its credentials? Today it is `user=username` which seems incorrect to me. In its current form, streaming doesn't work because `zms` does not get auth passed correctly

2. Other point: Does ZM recommend auth_relay to none - which you have answered

pliablepixels, 2018-11-29:
Ok, looks like @connortechnology and my posts crossed.

pliablepixels, 2018-11-29:
@connortechnology - I'll take this discussion to slack before I do a PR - have more Qs

commit e6b8a7bc (2018-11-29): resolves #2327
closed 2018-11-29 by knight-of-ni