ZoneMinder/zoneminder #2333: TimeZone API issue

mfauvain, 2018-12-06:
` function getTimeZone() {`
`    //http://php.net/manual/en/function.date-default-timezone-get.php`
`    $tz = date_default_timezone_get();`
`    $this->set(array(`
`      'tz' => $tz,`
`      '_serialize' => array('tz')`
`    ));`
`  }`

EDIT: as per @pliablepixels : PHP 7.2.13 (cli)

_Originally posted by @mfauvain in https://github.com/ZoneMinder/zoneminder/issues/2326#issuecomment-444887388_

pliablepixels, 2018-12-06:
To be clear so this issue is understood, your problem is that that your `getTimeZone` API is returning `UTC` and not the timezone you believe you have correctly configured in php (as you have not provided this information in your issue report)

Your PHP version is 7.2.13

Please do this:

**Step 1: Create a file called tz.php with this content**

```
<?php
echo "Output of date_default_timezone_get is:".date_default_timezone_get()."\n\n";

echo "This is because:\n";

echo "Output of first step:\n";
if (date_default_timezone_get()) {
    echo "date_default_timezone_set: " . date_default_timezone_get() . "\n";
}
else echo "data_default_timezone not set \n";

echo "Output of second  step: (5.4 and below only)\n";
if (isset($_ENV['TZ'])) {
    echo "TZ=" . $_ENV['TZ']."\n";
}
else {
    echo "TZ not set\n";
}

echo "Output of third  step: \n";
if (ini_get('date.timezone')) {
    echo "date.timezone: " . ini_get('date.timezone')."\n";
}
else echo "php ini date.timezone not set \n";

?>
```

**Step 2: Run it and post the output**

```
php -f ./tz.php
```

mfauvain, 2018-12-06:
@pliablepixels thank you vmuch for taking the time.
output of this function is : 
'Output of date_default_timezone_get `
`is:America/Mexico_City`

`This is because:`
`Output of first step:`
`date_default_timezone_set: America/Mexico_City`
`Output of second  step: (5.4 and below only)`
`TZ not set`
`Output of third  step: `
`date.timezone: America/Mexico_City`

pliablepixels, 2018-12-06:
Command line works fine, then.

What is the output of:
(After removing the manual hack)
`
https://yoursever/zm/api/host/getTimeZone.json
`

mfauvain, 2018-12-06:
gives (after removing the hack)
`{`
`    "tz": "UTC"`
`}`

pliablepixels, 2018-12-06:
Hmm so command line PHP works but the one linked to apache does not. Seems to me you have different PHP versions installed - is that true? It may be possible you have multiple ini files for multiple version and the one Apache uses is not the one you think its using?

pliablepixels, 2018-12-06:
So, for example, to make sure apache is using the version you think it is, I added this temporary API

```
function getInfo() {
   $inf = phpinfo();
   $this->set(array(
      'info' => $inf,
      '_serialize' => array('inf')
    ));
 }
```

Just add it before `function getTimeZone() {` in `HostController.php`.

Then do `https://yourserver/zm/api/host/getInfo.json` and see what is in the "Loaded Configuration File" row (4th from top) - is it what you are expecting?

connortechnology, 2018-12-06:
command line and apache php use different config files.  Indeed different versions of php will have different config files for each of those.  When upgrading on ubuntu from xenial to bionic, php 7.3 will be installed, but php 7.2 will remain the active version for apache2 until you tell it otherwise.

So it can be very confusing. We will need to verify which version of php apache is using. On ubuntu we can tell by looking in /etc/apache2/mods-enabled

pliablepixels, 2018-12-07:
So:

1. You have only 1 PHP version
2. CLI works fine and `date_default_timezone_get()` works fine
3. Apache is using the same version as the CLI 
4. For some reason, `date_default_timezone_get()` via Apache/PHP is failing and returning UTC

Not sure I have any more ideas. @connortechnology ?

Nocifer, 2018-12-12:
If you're on Arch and you're using the 'zoneminder' package from AUR, that means that there should be an /etc/php/conf.d/zoneminder.ini file with a timezone set inside, which overwrites whatever you may have set in your /etc/php/php.ini file. If this file is set to UTC, then that's what you'll get. The thing is, this file should NOT be set to UTC, it should have been set to your local timezone during ZM's setup. Since you said this was a brand new install, could it be that you installed ZM before you set the system's timezone? If that's the case then that could be your culprit.

For what it's worth, on my own system (Arch with the 'zoneminder' package from AUR), calling api/host/getInfo.json results in the correct timezone.

Nocifer, 2018-12-16:
Hey again, and sorry for the very late reply.

No, it's not mandatory to have ```date.timezone``` in there, it's just so users don't have to manually edit php.ini after they install ZM. The primary purpose of zoneminder.ini is to enable the extra PHP extensions needed by ZM, but it's really convenient to have everything ZM-related in the same place (and, according to ZM install instructions, setting the PHP timezone is important).

The thing is, any settings in conf.d/*.ini override their equivalents in php.ini, so it potentially can be a pain if the user is not aware of this functionality. With ```data.timezone``` it's obviously not an issue in 99% of cases, because your local timezone is whatever it is and that's that, no matter if it's configured in php.ini or zoneminder.ini or wherever else. But, well, your case is that remaining 1%: since you installed ZM before setting your timezone, zoneminder.ini pulled an incorrect UTC value and then PHP used that to override whatever you tried to set in the standard php.ini.

Or at least, that's what I'd *think* is the case. But you're saying that you were aware of zoneminder.ini and had even removed the offending line, correct? Or did you remove it *after* you'd already had issues with the incorrect timezone? What I mean to ask is: did removing that line give you the correct America/Mexico_City timezone instead of UTC?

In any case, this just made me realize that even though the AUR install script is supposed to be automated, I should probably still be printing info messages about any and all changes it makes (like installing zoneminder.ini and using it to set the PHP timezone) so the user is aware of them and can revert them or modify them without having to read the whole script to see how it works (which I always assumed was a given, but...).

closed 2019-04-18 by stale[bot]