ZoneMinder/zoneminder #2336: URI error with non-standard port

heegard, 2018-12-07:
I've been running ZoneMinder on Ubuntu for a couple of years. I'm currently on version 18.04 of Ubuntu. I've recently updated ZoneMinder to 1.32.2 using the package updater. I've discovered a new issue with this update.

The server works correctly on my local network.

The following method of accessing the server on the internet worked up-until this update.

On the internet, port 443 on mywebsite,com goes to another server on my local network. Thus, I have a port forwarded from port say 1234, to port 443. Thus I can access the ZoneMinder server like this:

https://mywebsite.com:1234/zm/index.php

This works and I can login. However, I cannot see the feeds. For example,

https://mywebsite.com:1234/zm/index.php?view=watch&mid=6

is broken. We can see the problem in the log:

web_js	ERR	SyntaxError: The URI is malformed.	https://mywebsite.com:1234/zm/tools/mootools/mootools-core.js

Viewing the generated webpage source:

<div id="imageFeed"><img id="liveStream6" src="https://mywebsite.com:1234:1234/zm/cgi-bin/zms?scale=100&amp;width=640px&amp;height=480px...

you can see the error (the port extension is repeated).

I suspect that someone familiar with some of the recent changes in the code could sort this out.

Please?

-Chris

PS: Zoneminder has improved substantially since I began, Thanks!

knight-of-ni, 2018-12-07:
This looks like what we fixed yesterday. see #2335

heegard, 2018-12-08:
That looks promising!

heegard, 2018-12-08:
I checked it out with my server, and had to make additional changes.

knight-of-ni, 2018-12-08:
We previously fixed the second use of splitting the port off HTTP_POST in this commit a1141d2dc43edd5bd89b79db6241c4258dc0faf2

heegard, 2018-12-08:
That version of Server.php seems to work.

Thanks!

closed 2018-12-08 by knight-of-ni