ZoneMinder/zoneminder #2339: (Reverse Proxies) URL generation troubles

abishai, 2018-12-09:
I think we really have some issues with reverse proxies.
Some non-critical urls are broken in web console (for example fps request).

There are 2 problems:
https://github.com/ZoneMinder/zoneminder/blob/master/web/includes/Server.php#L65
https://github.com/ZoneMinder/zoneminder/blob/master/web/includes/Server.php#L75

1. $_SERVER['HTTPS'] is not set and there are no way to set in in proxy configuration (it works as intended)
2. $_SERVER['SERVER_PORT'] is 80 (obviously)

To overcome this issue I set up prepend script for PHP-FPM
```
if(isset($_SERVER['HTTP_X_FORWARDED_PROTO']) && $_SERVER['HTTP_X_FORWARDED_PROTO'] == 'https'){
        $_SERVER['HTTPS']='on'; 
        $_SERVER['SERVER_PORT'] = '443';
}
```

Sure thing, the solution is awful and temporary.
We already have https://github.com/ZoneMinder/zoneminder/blob/17551eaceed33c821b0e3eb76676d01408f2ff51/web/includes/Monitor.php#L454 Probably, we can extend it to Server.php ?
And, we need port override.

connortechnology, 2018-12-11:
That line needs to be $url = $Server->UrlToApi().'/monitors/'.$this->{'Id'}.'.json';

This has been updated in master.

abishai, 2018-12-11:
On first page load for monitor I see a bunch of errors, for example Loading mixed (insecure) display content “http://xxxxxxxxxxxxx/cgi-bin/nph-zms?scale=50&width=1016px&height=768px&mode=jpeg&maxfps=30&buffer=1000&monitor=1&auth=dacc6765eb1f452876631d07628813da&connkey=34360&rand=1544556291” on a secure page
or http://xxxxxxxxxxxxxx/zm/index.php?view=request&request=status&entity=events&id=1&count=20&sort=Id%20desc

Stream is constantly restarting without mu hack with Detected streamWatch of type: stream stopped, restarting

abishai, 2018-12-11:
```
root@localhost [zm]> select * from servers;
Empty set (0.00 sec)
```
Should I register my server there and attach cameras to it ? If I understand Servers.php correctly, incorrect URL detection will be overrided in constructor.

connortechnology, 2018-12-11:
You can't mix http content on a https page.  All urls have to have https in front.  The outside world can't see your internal http reverse proxy.  They only know about the external https side.  

The code is SUPPOSED to default to whatever is currently in use.  So if you are accessing your site externally via https, then all url SHOULD have https as the protocol.  

You can use the Server entries to manually dictate how to reach a server, but the browser is going to reject any non-https content on an https page.

abishai, 2018-12-11:
Sure thing. In the first post I reported that link generation is broken and why it's happening behind proxy.

knight-of-ni, 2019-01-03:
We believe this issue was fixed in #2343. Feel free to create a new issue if this is not the case.

closed 2019-01-03 by knight-of-ni