ZoneMinder/zoneminder #2341: Invalid UE golomb code

jwtoler, 2018-12-11:
Since upgrading to 1.33.0 using the latest Git pull, my log file is filling up with “Invalid UE golomb code”.

connortechnology, 2018-12-12:
1.33.* is a fairly experimental development build.  We try to keep things relatively stable, but if scared, you may want to switch to the stable 1.32* release channel..

That being said, what you are seeing is a new feature where we are logging the output of the ffmpeg libraries.  The golomb code error is actually a bug in the ffmpeg decoding libraries.  I think in our next build we will demote errors from ffmpeg a bit, so this will become a warning instead of an error, but it will still be logged.

knight-of-ni, 2018-12-14:
See pr #2345 which allows the end user to turn off displaying these messages if they choose.

closed 2018-12-14 by knight-of-ni