ZoneMinder/zoneminder #2354: Changing timezone breaks just about everything

wdizzle, 2018-12-19:
In order to submit a bug report, please populate the fields below. This is required.

**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: PPA
- Ubuntu Server 18.04 LTS

**Describe the bug**
I realized after completely setting up 1.32.3 that my OS time was not correct. This causes the video feed and events to show the wrong time. I used dpkg-reconfigure tzdata to set it to the correct timezone (America/New_York) and then rebooted into the BIOS to set the hardware clock to the correct time. After rebooting back into the OS, Zoneminder would not load any video feeds at all. 

Apparently this is a known issue with AUTH_RELAY and the system time being off by a few hours,
therefore, I disabled this feature so the feeds would work again but all previous events are now unplayable.  Now the feed itself shows the correct time and the Events show the correct time on my desktop browser but ZMNinja on my phone shows the same event is off by -5 hours and the Log file in Zoneminder is off by +5 hours. On top of that enabling AUTH_RELAY causes the feeds to stop working entirely. 

[image]
[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Have an OS set to the wrong timezone, or change to another timezone
2. Use dpkg-reconfigure tzdata to set it to the correct time zone. 
3. Change the BIOS clock to match the timezone that you set. You can do this with the hwclock command as well. I don't think it makes a difference
4. Witness that enabling AUTH_RELAY breaks your streams, check your log and the time at the top be off by a number of hours. 
5. If you use ZMNinja, the event times will be off by a negative amount of hours instead of positive like the Log above.

**Expected behavior**
You should be able to change just ONE variable for time. I've spent 3+ hours trying to resolve what seems like such a simple issue at first only to come to the conclusion that I'll likely have to start from scratch and make damn sure the timezone and clock are set correctly before installing anything to the base OS. This would also make moving your server physically from one timezone to another a true PITA.

knight-of-ni, 2018-12-20:
We are looking into what we can do, but the core issue is that the timezone is recorded, independently, in three places:
- mysql (the timezone is set during initial install of the package)
- the operating system
- php.ini

All three of these must match, exactly, or things will break. I'm not sure from what you said if things are still broken, or if you figured it all out. If still broken, then check the timezone in mysql. That one is the least obvious of the three.

As far as "fixing" this, our options are limited since setting the timezone cannot be done by ZoneMinder. However, we can detect the condition and log something appropriate in the logs.

ASoTNetworks, 2018-12-20:
I have notice that when we switch to EST time from DST where we moved back 1 hour. The event for the overlapped hour started to have 2 MP4 files recorded into those events.

The only way to access those files is to browse to the directories of the events over SFTP.

But I wonder what about record in UTC and have the front end page adjust the timezone?

knight-of-ni, 2018-12-20:
Yes, storing everything in UTC and using localtime only when presenting information to a human is exactly what we have in mind. However, there will still be breakage if all three timezones do not match up exactly.

wdizzle, 2018-12-21:
Knight-of-ni, I ended up rebuilding the VM from scratch to fix the issue but I still have the broken VM to attempt any suggestions/fixes. Oddly enough, I set the time in mysql but php.ini never crossed my mind. I'll look into setting it there and see if it resolves the issue.

I do agree that some kind of conditional check and alert to the oddity would be nice. This didn't seem to be an issue in 1.30. I made the same error when setting it up and remember just doing the dpkg-reconfigure tzdata to fix it. 

At least this frustration finally prodded me into fixing the dang timezone on my hypervisor.

knight-of-ni, 2018-12-21:
An incorrect timezone has broken streaming since the hashed option has been available with AUTH_RELAY.... so a long time.

If you search the user forum, generally speaking, anyone who reports they had to set AUTH_RELAY to plain, rather than hashed, most likely has one of the three timezones set incorrectly. 

It's easy to do. All one has to do is forget to set the system timezone before installing mysql.

Ubuntu has multiple php.ini's. Make sure you edit the right one. See the installation documentation to verify which one you should update.

closed 2019-01-15 by connortechnology

connortechnology, 2019-10-31:
are you on 1.33.14? If so, we moved this to Options -> System -> Timezone.

giuarb, 2019-10-31:
Hi Connor, I saw your announcement for 1.33 but I am on still on 1.32.3
any suggestion?

connortechnology, 2019-10-31:
Yes, update to 1.33.14

giuarb, 2019-10-31:
thank you Connor, do I use your https://launchpad.net/ubuntu/bionic 1.33.14~20191023105216-bionic
I am on ubuntu server 18.04

connortechnology, 2019-10-31:
Yes.
sudo add-apt-repository ppa:iconnor/zoneminder-master
sudo apt-get update
sudo apt-get install zoneminder

giuarb, 2019-10-31:
will it just upgrade from 1.32 ?

PR #2387 (2019-01-15): implement timezone check function
fixes #2354 
supersedes #2378 

This implements a timezone check function that compares the system timezone to php and mysql timezones. It does this in a way that is compatible with any Linux or BSD type system without looking at specific file paths on the filesystem. 

If a discrepancy is found between the three timezones, an appropriate error is logged to alert the end user to fix the problem.

This is meant as a short term solution to #2354. A redesign of how times are stored in ZoneMinder is planned for a future release.