ZoneMinder/zoneminder #2355: The video could not be loaded

ozzy1024, 2018-12-19:
**Describe Your Environment**
- Version of ZoneMinder [1.32.2]
- How you installed ZoneMinder RPMFusion]
- Full name and version of OS: Centos 7 (7-6.1810.2.el7.centos.x86_64)

**Describe the bug**
Randomly, playback of some events results in a black screen with this error:  The video could not be loaded, either because the server or network failed or because the format is not supported.  I have  5 cameras (4 Hikvision and 1 Foscam), and the issue is not specific to one camera.  

I did some research through forums both on ZM and others, and it appears this is not a new issue.  

**To Reproduce**
I have not been able to reproduce the issue on demand.  It seems random.

**Expected behavior**
The event is supposed to play.
[image]

**Debug Logs**
I'm a newbie here with ZM, and my Linux skills are not expert level.  Let me know what you need, and I'll get it.

knight-of-ni, 2018-12-20:
This is a new problem, it is not widespread, and please don't double post. The problem is in your browser. 
It does not support the h264 codec. This is specific to browsers shipped with redhat distros. Try switching browsers. If you want more details, feel free to discuss this in the user forum or our slack channel.

closed 2018-12-20 by knight-of-ni

ozzy1024, 2018-12-20:
I've tried both Chrome and Firefox, and the behavior is the same.  Why would it play some video and not others if it's a codec issue?