ZoneMinder/zoneminder #2358: Invalid UE golomb code

zhuykovkb, 2018-12-21:
I broke my production with one of the latest update.

Version https://github.com/ZoneMinder/zoneminder/commit/c4bf67b06371a2510d083c8b7e1bc81b9df4b7b1 worked perfectly. But i decided to upgrade my zoneminder to the latest commit 6eb7892fc758a098f6ecd0b62ae505b9744691e2 and all cameras goes down (see attachment). In the LOG i see many WARN like as "Invalid UE golomb code"(see attachment).
Help me, please =)

I build zm with that  [Dockerfile](https://github.com/ZoneMinder/zmdockerfiles/blob/master/development/ubuntu/xenial/Dockerfile).
[image]
[image]

zhuykovkb, 2018-12-21:
i'm afraid the problem lies elsewhere. I've opened brower console and see, that all links on the page points to http proto instead of https (see attachment).
[image]

tell me, please, how zm understand which protocol use a client? i'am using nginx before apache zoneminder, maybe i forgot to set header in proxy request?
[image]

connortechnology, 2018-12-21:
When figuring out whether to use http or https, we look first for if we are using https, then in the HTTP_X_FORWARDED_PROTO header.

Alternatively, you can enter a Server in options->Servers and assign it to your monitors and it will use whatever you enter in there.

knight-of-ni, 2018-12-21:
As a side note, see #2341 if you don't want to see those golomb code messages in your logs.

zhuykovkb, 2018-12-24:
hmm..okey, i set ZM_LOG_FFMPEG to 0 and no more lines in log. ok. but i have the trouble with using ssl. why all links on the page points to http proto instead of https and port 80, instead of 443?
[image]

zhuykovkb, 2018-12-27:
> When figuring out whether to use http or https, we look first for if we are using https, then in the HTTP_X_FORWARDED_PROTO header.
> 
> Alternatively, you can enter a Server in options->Servers and assign it to your monitors and it will use whatever you enter in there.

okey. can you explain why links, look at previous my post, in the firebug (browser dev tool) points to http://zm.beget.ru:80 instead of httpS://zm.beget.ru?

connortechnology, 2018-12-27:
I believe I already did. You aren't setting a header correctly.  You can work around it by explicitly setting it in a server entry.

zhuykovkb, 2018-12-27:
> I believe I already did. You aren't setting a header correctly. You can work around it by explicitly setting it in a server entry.

i'm sorry, but i don't think i'm setting a header correctly. I made the screenshot in debug mode
[image]

can u tell me what i'm doing wrong?

zhuykovkb, 2019-01-09:
i solved my problem.
```
proxy_set_header X-Forwarded-Port 443;
```

closed 2019-01-09 by zhuykovkb