ZoneMinder/zoneminder #2363: Zones types, Preclusive and Inclusive need to be reworked.

aaronsta1, 2018-12-25:
Currently the zone types Preclusive and Inclusive does not work as intended.

the intended use as per the wiki.
https://wiki.zoneminder.com/Defining_Zones
states the preclusive zone should stop alams from happening if its triggered first.. the problem is IF its triggered first.
as per iconner in this post
https://forums.zoneminder.com/viewtopic.php?f=38&t=27828
he states the preclusive zone is not scanned first.. its scanned near last. so that means if you have 2 zones and you want to mask motion in one zone by using a preclusive zone, it will almost NEVER work. 
example of what you want to use preclusive for.. say you have a driveway and you want to keep car headlights from triggering the alarm but you want the alarm zone sensitive enough to detect people.
you would put the active alarm over the region you want to sense motion, and you would put a small strip of preclusive zone up top towards the street.. this way if a car enters the driveway you SHOULD trigger the preclusive zone first.. then no alarm until the car enters your driveway and shuts the lights off.

also there is a bug that if it does work, the countdown timer stops as soon as the alarm stops, not the duration you tell it in the extend alarm frames spot.

as per the wiki 
https://wiki.zoneminder.com/Defining_Zones
the inclusive zone should extend an alarm if the alarm is triggered first.
this means if you put an alarm zone on say at the beginning of a path, with an inclusive zone on the rest of the path, if someone walks down that path and triggered the alarm and then the inclusive as per the wiki the alarm should keep going.. but it does not. as soon as the person leaves the active zone, the alarm stops. the inclusive zone seems to be completely ignored.. at least i could never get it to work.

i am using version 1.32.3 from iconner ipa on ubuntu 18.04 with a PC that has a I5 4660 with 12 GB of memory and a 2 TB purple WD hard drive.
i have 4 cameras.
2 of my cameras use the RTSP source with ffmpeg with a resolution of 1280x960 5FPS and 2 use the REMOTE HTTP source with a resolution of  640x480 5FPS.

knight-of-ni, 2018-12-25:
This is really two reported issues.
We will look into it the first reported issue, concerning preclusive zones. I'm not sure if it has always been processed last or something got bumped more recently. 

The second reported issue, regarding inclusive zones, is not the right use case. I have responded to that in the user forum. Please keep that part of the discussion there.

knight-of-ni, 2018-12-28:
@aaronsta1 Have you tried setting the Extent Alarm Frame Count?

Documentation for that can be found at the bottom of this page:
https://zoneminder.readthedocs.io/en/latest/userguide/definezone.html#basic-parameters

It might help.

I've been speaking with @connortechnology about this, and I believe he was mistaken about the order.

Preclusive zones do indeed get checked before Active and Inclusive zones are checked:
https://github.com/ZoneMinder/zoneminder/blob/master/src/zm_monitor.cpp#L2630

There may still be something going on that is not right, but it does not appear to be the order.

Please experiment with the Extend Alarm Frame count then report back.

knight-of-ni, 2018-12-28:
Forgot to mention, set your Debug to level 3. It will tell you precisely when the preclusive zone activates.

aaronsta1, 2018-12-28:
ill try the debug but ive proven it doesnt work because i set the whole screen an active and then the preclusive on top of it. which should zero out the active. then i set the extend frame count to 100 which is a lot.. it never counts down. 

ill try the debug and see.

aaronsta1, 2018-12-28:
ok here we go.. it shows the "driveway" zone saw motion, but the preclusive did not. 
```
12/28/18 12:39:13.274529 zma_m3[10473].DB3-zm_monitor.cpp/2624 [Checking preclusive zone New - old score: 0, state: quiet]
12/28/18 12:39:13.274640 zma_m3[10473].DB3-zm_monitor.cpp/2662 [Checking active zone Driveway]
12/28/18 12:39:13.277970 zma_m3[10473].DB1-zm_zone.cpp/731 [Driveway: Pixel Diff: 115, Alarm Pixels: 39196, Filter Pixels: 38284, Blob Pixels: 38187, Blobs: 1, Score: 15]
12/28/18 12:39:13.277993 zma_m3[10473].DB3-zm_monitor.cpp/2667 [Zone is alarmed, zone score = 15]
12/28/18 12:39:13.278000 zma_m3[10473].DB3-zm_monitor.cpp/1399 [After motion detection, last_motion_score(0), new motion score(15)]
12/28/18 12:39:13.278007 zma_m3[10473].DB3-zm_monitor.cpp/1445 [Detected new event at (1546029553.259712)]
```

aaronsta1, 2018-12-28:
it does seem to be working, just not all of the time.
```
12/28/18 12:51:58.552397 zma_m3[11036].DB3-zm_monitor.cpp/1345 [Motion detection is enabled signal(1) signal_change(0)]
12/28/18 12:51:58.552730 zma_m3[11036].DB3-zm_monitor.cpp/2624 [Checking preclusive zone New - old score: 4, state: quiet]
12/28/18 12:51:58.556257 zma_m3[11036].DB1-zm_zone.cpp/731 [New: Pixel Diff: 55, Alarm Pixels: 19878, Filter Pixels: 16581, Blob Pixels: 12690, Blobs: 1, Score: 4]
12/28/18 12:51:58.556282 zma_m3[11036].DB3-zm_monitor.cpp/2629 [Zone is alarmed, zone score = 4]
12/28/18 12:51:58.556295 zma_m3[11036].DB3-zm_monitor.cpp/1399 [After motion detection, last_motion_score(0), new motion score(0)]
12/28/18 12:51:58.556307 zma_m3[11036].DB3-zm_monitor.cpp/1445 [Detected new event at (1546030318.551573)]
```

aaronsta1, 2018-12-29:
i see a typo in the log.. wonder if that has anything to do with it.
12/28/18 18:34:40.703161 zma_m3[30170].DB3-zm_monitor.cpp/2635 [Preclusive Zone New alarm Ends. PrevÃ­ous score: 0]

notice previous.. its spelled wrong.. with your log too.

aaronsta1, 2018-12-29:
i think maybe i might have a plausible explanation but i dont know how to read the code to see..
so i put the preclusive to 1 pixel change, which should basically put it in a constant alarm.
i still get a few notifications on zmninja when cars go by.. 
here is what i think is happening.. 

when the extended timer hits 0, instead of starting over to reread the preclusive zone to restart the timer, it continues and goes to the alarm. doing so activates the alarm and sends me a notification.

im almost certain this is what is happening because the alarm is usually right after it hits 0.

aaronsta1, 2018-12-31:
so after watching the logs for the last couple days i think i am correct on my assumption of whats happening.

no matter what you put the extend alarm frames at. as soon as they count down to 0 if there is still motion it causes an alarm.
you can test.
put the camera in front of a fan, set the alarm to something that the fan motion triggers. so its constantly being alarmed.

make the active and preclusive the same, and let it go.. after awhile you will see alarms being triggered when they shoudlnt be.

knight-of-ni, 2018-12-31:
@aaronsta1 what I need from you is a more quantitative description of how you have this set up. When I get home from the office, I'll set it my zones exactly the same as yours. In order to do that I need to know *exactly* what you entered into the zone fields. The simplest thing to do would be to screenshot the zone edit screens. I need to see all the parameters you entered, not just some of them.

I'd also like to see your Buffers tab as well since that effects the potential length of an alarm.

aaronsta1, 2019-01-01:
here is how i have the camera setup.
its looking out at the street, i dont really have any issues during daytime, but at night the cars headlights shine in the driveway when they make a left at that stop.
the top is preclusive, its set to 80. according to the dubug, the headlights are around 100 to 120.
the bottom is active, its set to 25. this is enough to detect people walking through.

my issue is, the cars while they do trigger the preclusive zone, it also triggers the active zone.

[image]
[image]
[image]
[image]

aaronsta1, 2019-01-01:
here is a frame of an alarm i just got. so you can see my example, it should not have alarmed.
[image]

knight-of-ni, 2019-01-01:
No, it isn't broken. Check your alarm frames in the active zone. They the motion for your alarm frames after the first alarm frame will be outlined.
In hindsight, you won't be able to see the red outlines in Preclusive mode since as soon as it triggers, it clears the alarm.

aaronsta1, 2019-01-02:
ok that was really helpful, should be enabled by default.
i checked the preclusive zone by making it active and setting the active zone to inactive.
the preclusive zone is working, it detects the cars headlights and outlined them with red.
i still dont know why its not working.

knight-of-ni, 2019-01-02:
Setting analysis frames on by default is a bit like having debug on all the time. It consumes resources, which is why it is off by default. The project is also moving away from jpeg generation. Maybe someday we will be able to just overlay the blob on top of the mp4 stream. I do agree that we may not need the global variable under Options anymore, now that one can enable or disable analysis frames with each monitor.

I still can't get a false trigger on my demo system. I've configured the zones to match your screenshots exactly. I'll continue to work on it.

Here are some things to try on your end:
- What happens when you lower the pixel threshold of the preclusive zone to something less than 80? Say 25?
- Rather than use a preclusive and an active zone, try changing the Active zone to an Exclusive zone and change the Preclusive zone to an Active zone. That should give you events, only when there is activity in your driveway and no other zone. Keep Blob method enabled in both zones so you can see the red blobs.

knight-of-ni, 2019-01-02:
I have added removal of the global CREATE_ANALYSIS_IMAGES to our project task list since we can enable this on a per-monitor basis now:
https://github.com/ZoneMinder/zoneminder/projects/2

aaronsta1, 2019-01-02:
the problem is the driveway gets lit up like this. it completely ignores the preclusive zone.
if i set the preclusive zone to more sensitive, then the cars that go by during the day set it off and its just being constantly alarmed.
[image]

aaronsta1, 2019-01-02:
ill try the exclusive zone like you mention.

aaronsta1, 2019-01-02:
yeah i almost never get an alarm during the day, even without the preclusive zone.

is there a way to change the settings for day/night? if so i can just set the active zones threshold to like 40 and i wont get any detections from reflections but it will still detect people at night because of the ir.

aaronsta1, 2019-01-02:
ok so i did do a test, like yours. i set a webcam up in a dark room set to modect and put the active and preclusive ontop on the default setting, 25/3/3/2 and i turned on and off the lights.. and it works.

i should mention that the outside cam is set to mocord. i wonder if that is affecting anything.

knight-of-ni, 2019-01-02:
Absolutely. You can multiple (many) run states that are completely different from one another: different sensitivity, different types of zones, different number of zones, etc. Once the run states are defined, the simplest way to changes states is via cron. 

However, if you want to full nerd, you could wire a day/night sensor to an arduino, which would send an API command over wifi to switch run states when the sun sets and then rises again.... I think I just gave myself a new project to do. :-)

Runstates are mentioned here: https://zoneminder.readthedocs.io/en/stable/userguide/gettingstarted.html?highlight=runstates

The paragraph covering runstates has a link to the FAQ and to an interesting project where the user used the presence of their mobile device to change runstates.

aaronsta1, 2019-01-02:
the runstates look interesting.. perhaps you could add a time in the ui with a dropdown so i can set day/night run states without changing them manually? 

hmm i wonder if this can be added to zmninja..

aaronsta1, 2019-01-03:
so i guess its not a issue with the preclusive zone itself, but i think you are right. from my testing if i say put the extend alarm to 99 which is several seconds when the car stops at the stop sign the countdown stops at whatever its at when the alarm runs out. so in order to use the preclusive zone, you have to set the buffer to around the same time as you put the extend alarm.

altho it seem sometimes the zone ignores bright lights. 
here is a good example of what i just saw, it ignored the bright light and just focused on the reflection from the driveway. not certain how to fix that.

[image]

aaronsta1, 2019-01-03:
for future versions it would be helpful if you added the pixel_diff and alarm area to the notes section of the alarm.

knight-of-ni, 2019-01-03:
Agreed. Presenting the right amount of information to the end user in an easier to understand format is definitely something we want to improve on. Bringing the pxel_diff and alarm area forward is a good idea.

ZoneMinder works by measuring the RGB color difference of each pixel to the same pixel in the frame before it. It does not count bright lights as motion when that light is unchanging between frames. It only counts the change from dark to light, or vice versa. When the pixel stops changing color, it is no longer counted as motion. 

The screenshot you posted is why you want a longer post roll. You want the extended alarm frame count to still be in effect at this point in the event. The top left corner is all white, and so was the the frame before it, hence no motion is counted. As the light spreads out and begins to fade, ZoneMinder can see the pixels changing so it counts that.

Note that if the camera had a wider dynamic range, the top left corner would not be quite as washed out and ZoneMinder would have a better time making out pixel changes.

aaronsta1, 2019-01-03:
i think i understand how it works. i didn think to change the post buffer or the blending because i was thinking the preclusive skipped all of that and just went to the extend frame timer. which i thought should work as i could set it to say 5 seconds and by then the car would have been out of the frame.. i was getting frustrated because i didnt understand why it wasnt working.

thanks for letting me know about the profiles, i had no idea you could do that. i can make a script to get the sunrise and sunset time and just have it switch automatically from day to night profile.

ill go ahead and close this.

closed 2019-01-03 by aaronsta1