ZoneMinder/zoneminder #2364: Setting Alarm Frame Count is not affecting events

LNFWebsite, 2018-12-25:
ZM 1.32.3
Ubuntu 18.04 LTS

When setting `Alarm Frame Count` on cameras, I'm still getting events with 1 alarm frame within them. I've tried multiple values to no avail. This feature worked fine before upgrading from ZM 1.30, yet seems to be broken in 1.32.

Thanks

connortechnology, 2018-12-27:
What are your pre-event counts and analysis fps settings?

knight-of-ni, 2018-12-28:
I am unable to duplicate this. I set alarm frame count to 3, on a camera pointing right at me.
I let it run for a couple hours and did not see any events with less than 4 alarm frames, even after intentionally trying to reproduce the problem with varying degrees of movement.

Please provide all the details of the General, Source, and Buffers tab for the camera in question. Screenshots are preferred.

bceylon, 2018-12-28:
I’ve seen this alot and was about to report it myself. Happens with two outdoor cameras with ffmpeg source type and h264 passthrough. Other settings:

Maximum & Alarm Maximum fps = 8
Analysis fps = empty
Image buffer size = 20
Pre-event count = 5
Post-event count = 50
Alarm count = 2 

Tried now to reproduce but got way too many alarm frames, happens mostly during night with IR-leds and rain/snow. Any particular logs that would be usefull?

LNFWebsite, 2019-01-02:
Hi there, sorry about the late reply.

My system is two RTSP camera streams both set to 5 FPS (on camera).

I would set a pre-event image count, but am having #2361 issue.

Here's the config for one of them, the same alarm frame count issue occurs on both:

[image]
[image]
[image]

knight-of-ni, 2019-01-03:
Try removing the values from the max fps fields.  In older versions of zm, placing a value that was *higher* (not equal to) the frame rate from the camera was a workaround to keep the cpu from exploding should the camera go offline. Isaac has fixed that. 

If you have a value that is equal to or less than the frame rate from the camera, then ZoneMinder monkeys with the frames, rather badly so this might be the cause. We are going to rework the fields In a future version of ZoneMinder so these fields are not even shown in most cases.

LNFWebsite, 2019-01-03:
Okay, I've removed those values and will let you know if there's any change.

I've always set those values because ZM seems to report a framerate that wanders pretty significantly around the actual 5 FPS of the camera.

knight-of-ni, 2019-01-03:
A fluctuating frame rate is a camera problem. I have had cameras that do that too. 

I just want to make sure we are clear. ZoneMinder does not have the ability to lower the framerate of any incoming stream. The use case you are using those fields for is precisely what you should not do. 
Those fields were only designed for non-streams, like snapshots or reading a local jpg off the filesystem.

This is right out the help text:
> Failure to adhere to these limitations will cause a delay in live video, irregular frame skipping, and missed events

If frames are being lost, that could easily skew the alarm frame count.

LNFWebsite, 2019-01-05:
Maybe so, however this issue is not related to the set frame rate in ZM, as some events still have less than the Alarm Frame Count set (observed since settings change and restart). In previous versions of ZM I have not had any issue related to setting the FPS in ZM even though this should cause undesirable side effects.

knight-of-ni, 2019-01-05:
Neither Isaac nor I can duplicate the problem. For experimentation sake, place a value of 15 in the pre event count. On my end, I'll put a value of 0 and see if it makes a difference.

knight-of-ni, 2019-01-07:
Today I setup three cameras with the exact same settings as yours. With a pre event count of 0, I can duplicate the problem. I then switched the pre event count to 1, and the problem appears to have gone away. I will probably place a minimal value restriction on the pre event image count field to be either 1 or to match the alarm frame count, pending further tests.

LNFWebsite, 2019-01-07:
Okay, great! Thanks I'm going to switch mine to 1 to see if that fixes it for me. Setting the pre event count to 15 causes the other issue I referenced regarding video file timing being corrupted and makes it impossible to view event videos. I'll see if it does it at 1...

closed 2019-03-15 by stale[bot]