ZoneMinder/zoneminder #2375: Docs reference specific PHP version and file location on Ubuntu Installation instructions regarding the PHP config file

SalmonMode, 2019-01-01:
I noticed in the docs [here](https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html), that in step 9 it references a specific version of PHP and a specific location for the php config file, but some might not have that version of PHP, or their PHP may be loading a different config file.

This might throw some people off, so to help them (in addition to having more correct and thorough documentation), this step in the documentation should also describe how to find the config file that PHP will actually be loading.

knight-of-ni, 2019-02-11:
From the context of your message I am guessing you are running Ubuntu 18.04. Notice the documentation you referenced is for Ubuntu 16.04.

The version of php is tied to the version of the linux distro it ships with. Centos 7 ships with php 5.4, fedora 29 php 7.2, ubuntu 16.04 php 7.0, ubuntu 18.04 php 7.2, etc. These typically do not change by the software manufacturer because the chance to cause breakage is significant. 

The path to the php.ini file is correct for an Ubuntu 16.04 system. That's not to say we should not update the documentation at all. However what should be done, instead, is a new set of installation installation instructions written specifically for Ubuntu 18.04.

SalmonMode, 2019-02-11:
Yeah, you're correct. That was the version of Ubuntu I was working with. I'll modify my PR to just add a new section for 18.04 later today.

SalmonMode, 2019-02-12:
Fixed up the PR :+1:

PR #2376 (2019-02-12): Fixes #2375
Fixes #2375 

I noticed this because my install of PHP was referencing a config file in `/etc/php/7.2/cli/`, instead of `/etc/php/7.0/apache2/`.
commit f7b31c89 (2019-02-12): Fixes #2375 (#2376)

* updated docs to include instructions on how to find the loaded PHP config file

* Added note about verifying timezone change

* revert docs back to master branch's content

* added installation guide for Ubuntu 18.04

The only difference between Ubuntu 16.04 and 18.04 is the version of
PHP that comes installed, which changes the location of its config file

* reverting debian instructions to master
closed 2019-02-12 by connortechnology