ZoneMinder/zoneminder #2381: Can't add 'WebSite' Monitor type

jacobalberty, 2019-01-03:
**Describe Your Environment**
ZoneMinder 1.32.3 in the development dockerfile
Host os is Debian Stretch docker environment is ubuntu xenial

**Describe the bug**
Can't add a 'WebSite' monitor.
I did some investigation by dumping my schema with mysqldump -d Looking at the `Monitors` table it appears the enum for the `type` column is missing the 'WebSite'  type:

```
  `Type` enum('Local','Remote','File','Ffmpeg','Libvlc','cURL') NOT NULL DEFAULT 'Local',
```

This was a recent install so it appears the new install definition script is missing the full list of types for `Monitors`, and `MonitorPresets` tables as the `Controls` and `ControlPresets` contain the full list including 'WebSite'

**To Reproduce**
1. Click add monitor
2. Change source type to 'Web Site'
3. Change function to 'Monitor'
4. Enter web site url and width/height in source.
5. Click save

**Expected behavior**
I expect it to save the monitor, instead the dialog stays up and website does not get saved

**Debug Logs**
```

01/02/19 19:56:42.201600 web_php[20922].ERR [SQL-ERR 'SQLSTATE[01000]: Warning: 1265 Data truncated for column 'Type' at row 1', statement was 'INSERT INTO Monitors SET
 `LinkedMonitors` = NULL, `Name` = 'Monitor-5', `ServerId` = NULL, `StorageId` = '1', `Type` = 'WebSite', `Function` = 'Monitor', Enabled = 1, `RefBlendPerc` = '6', `Al
armRefBlendPerc` = '6', `AnalysisFPSLimit` = NULL, `MaxFPS` = '30', `AlarmMaxFPS` = '30', `Triggers` = '', `Device` = '/dev/video0', `Channel` = '0', `Format` = '255',
`Palette` = '0', `V4LMultiBuffer` = NULL, `V4LCapturesPerFrame` = '1', `Protocol` = NULL, `Host` = NULL, `Port` = '80', Method = '', `Options` = NULL, `SaveJPEGs` = '0'
, `VideoWriter` = '1', `EncoderParameters` = '# Lines beginning with # are a comment \r\n# For changing quality, use the crf option\r\n# 1 is best, 51 is worst quality\
r\n#crf=23', RecordAudio = 0, RTSPDescribe = 0, `LabelFormat` = '%N - %d/%m/%y %H:%M:%S', `LabelX` = '0', `LabelY` = '0', `LabelSize` = '1', `ImageBufferCount` = '20',
`WarmupCount` = '0', `PreEventCount` = '0', `PostEventCount` = '5', `StreamReplayBuffer` = '0', `AlarmFrameCount` = '1', Controllable = 0, `ControlId` = NULL, `ControlD
evice` = NULL, `ControlAddress` = NULL, `AutoStopTimeout` = NULL, TrackMotion = 0, `TrackDelay` = NULL, `ReturnLocation` = '-1', `ReturnDelay` = NULL, `EventPrefix` = '
Event-', `SectionLength` = '600', `FrameSkip` = '0', `MotionFrameSkip` = '0', `AnalysisUpdateDelay` = '0', `FPSReportInterval` = '100', `DefaultView` = 'Events', `Defau
ltRate` = '100', `DefaultScale` = '100', `WebColour` = 'red', Exif = 0, `SignalCheckPoints` = '10', `SignalCheckColour` = '#0000c0', `Path` = '<website address>', `Width` = '640', `Height` = '480', `Refresh` = NULL, Sequence = 4' params:] at includes/database.php line 158
01/02/19 19:56:42.283770 web_php[20922].ERR [Error saving new Monitor.] at includes/actions.php line 568

```

jacobalberty, 2019-01-03:
So looking at it it looks easy enough to change db/zm_create.sql.in to have the correct structure for those two tables and that would fix it for new installs, but I guess existing installs need a new zm_update-<version>.sql and I know nothing about creating those, do I just create a zm_update-1.32.4.sql and put the change statements in there? or does it need to be referenced somewhere?

knight-of-ni, 2019-01-03:
I'm pushing a pr now

PR #2382 (2019-01-03): add WebSite type to enum
fixes #2381
closed 2019-01-03 by connortechnology