ZoneMinder/zoneminder #2388: Run States not working as you said

aaronsta1, 2019-01-05:
not sure where to put this so i made a new issue.
i made 2 run states. day/night
i saved each one with its own preclusive zone and settings.

when i load them, it doesnt change anything.
i even hit restart and apply after.

the only thing a run state seems to do is set the monitor state.. i can change it from mocord to record. but i cant set up a new zone and have it carry over.

aaronsta1, 2019-01-05:
also it seems if you restart the pc the run state goes to the first one, not the one you had it set to.

knight-of-ni, 2019-01-05:
Yeah, I had forgotten an additional step is required in our previous conversation. 

Searching the user forum reveals a user who asked the same question and got an answer: https://forums.zoneminder.com/viewtopic.php?t=20699

Instead of opening a new bug report in our github forum each time you run into an issue you can't immediately solve, please start asking questions in our user forum and/or our slack channel. Slack is great for getting answers to quick questions. Someone is always around during normal US business hours. You can even ping me directly if you want.

closed 2019-01-05 by knight-of-ni