ZoneMinder/zoneminder #2390: PHP output_buffering set to Off breaks media-related views

davidjb, 2019-01-05:
**Describe Your Environment**
- Version of ZoneMinder: v1.32.3
- How you installed ZoneMinder `pkg install zoneminder`
- Full name and version of OS:  FreeBSD 11.2

**Describe the bug**
Certain views fail to display their media content when PHP's `output_buffering` is set to `Off` (which is the [default for PHP](https://secure.php.net/manual/en/outcontrol.configuration.php).  Packaged versions of PHP usually set this option to on in `php.ini` configs they provide, but I came across this issue as no default php.ini is provided in FreeBSD (templates are provided, but it's up to you to set one up).  

Without output buffering on, notices and warnings are outputed to the the HTTP response (see below).  This has the effect of injecting HTML text into responses for videos and images, causing them to be invalid.  The specific views and their error messages are below.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Console
2. Click on an event count
3. Note that thumbnails are broken
3. Click an Event
4. Note that video playback is broken

**Expected behavior**
Either output buffering to be optional (eg by handling the notices/warnings thus preventing the appearance of the HTML), or make a requirement to be set in php.ini.  In the case of the latter, just some explanation of the requirement in the docs would be sufficient (eg as there is with timezone setup).

**Debug Logs**

These views include:

1. Event video viewing or downloading: `/zm/index.php?mode=mpeg&format=h264&eid=000&view=view_video`.  Error message is attached to the top of an MP4 video:
   ```html
    <br />
    <b>Notice</b>:  ob_clean(): failed to delete buffer. No buffer to delete in <b>/usr/local/www/zoneminder/views/view_video.php</b> on line <b>101</b><br />
    [rest of binary MP4 content]
   ```

1. Event thumbnail or frame viewing: `/zm/index.php?eid=000&fid=snapshot&view=image&width=128&height=72`
    ```html
    <br />
    <b>Notice</b>:  ob_end_clean(): failed to delete buffer. No buffer to delete in <b>/usr/local/www/zoneminder/views/image.php</b> on line <b>251</b><br />
    <br />
    <b>Warning</b>:  Cannot modify header information - headers already sent by (output started at /usr/local/www/zoneminder/views/image.php:251) in <b>/usr/local/www/zoneminder/views/image.php</b> on line <b>252</b><br />
    <br />
    <b>Warning</b>:  Cannot modify header information - headers already sent by (output started at /usr/local/www/zoneminder/views/image.php:251) in <b>/usr/local/www/zoneminder/views/image.php</b> on line <b>287</b><br />
    [rest of binary JPEG content]
    ```

1. Exporting event as archive (eg zip):  `/zm/index.php?view=archive&type=zip`
    ```html
    <br />
    <b>Notice</b>:  ob_clean(): failed to delete buffer. No buffer to delete in <b>/usr/local/www/zoneminder/views/archive.php</b> on line <b>48</b><br />
    <br />
    <b>Warning</b>:  Cannot modify header information - headers already sent by (output started at /usr/local/www/zoneminder/views/archive.php:48) in <b>/usr/local/www/zoneminder/views/archive.php</b> on line <b>49</b><br />
    <br />
    <b>Warning</b>:  Cannot modify header information - headers already sent by (output started at /usr/local/www/zoneminder/views/archive.php:48) in <b>/usr/local/www/zoneminder/views/archive.php</b> on line <b>50</b><br />
    <br />
    <b>Warning</b>:  Cannot modify header information - headers already sent by (output started at /usr/local/www/zoneminder/views/archive.php:48) in <b>/usr/local/www/zoneminder/views/archive.php</b> on line <b>51</b><br />
    [rest of binary ZIP content]
    ```

connortechnology, 2019-01-07:
One fix for this is setting Options -> Logging -> LOG_LEVEL_WEBLOG to None.  

This setting would really only be for developers and really isn't that useful anyways.  I personally think we should remove it.

davidjb, 2019-01-08:
@connortechnology My LOG_LEVEL_WEBLOG is already set that way, which was the default for my install.  I should have mentioned these outputs are PHP errors so their display is controlled by `display_errors` in php.ini.  Turning this option to `Off` hides the errors, thus making the media views work, but it doesn't resolve the errors from happening.  

The PR #2395 looks like it'll solve the first two issues, but needs to have this line (https://github.com/ZoneMinder/zoneminder/blob/c28fd9cad06e1d6d72eee14273da3cdff60d76a6/web/views/archive.php#L48) removed as well to fix the archive downloader

connortechnology, 2019-01-09:
I've added it to the pr

PR #2395 (2019-02-22): remove ob_clean stuff which logs errors when output buffering is turned off
fixes #2390 

We put these in to solve problems with extra bytes being output.  Technically they should not be needed.  I think the extra bytes were from the WEBLOG logging.  

I think we need to get rid of WEBLOG Logging.

We might need to discuss these changes further.
closed 2019-02-22 by connortechnology