ZoneMinder/zoneminder #2394: "Unable to load storage area for id" when viewing monitors

Warudo, 2019-01-06:
- Version of ZoneMinder [release version, development version, or commit]: v1.32.3
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]: zoneminder package in Arch Linux AUR
- Full name and version of OS: Arch Linux

When I view monitors (single or in montage mode), I get the following errors in my log:
```
2019-01-06 16:07:55 | zma_m4 |   | 16642 | ERR | Bogus number of lines return from query, 0 returned for query SELECT Id, Name, Path, Type, Scheme from Storage WHERE Id=1. | zm_db.cpp | 129
2019-01-06 16:07:55 | zma_m4 |   | 16642 | ERR | Unable to load storage area for id 1: | zm_storage.cpp | 69
2019-01-06 16:07:54 | zms_m4 |   | 16638 | ERR | Bogus number of lines return from query, 0 returned for query SELECT Id, Name, Path, Type, Scheme from Storage WHERE Id=1. | zm_db.cpp | 129
2019-01-06 16:07:54 | zms_m4 |   | 16638 | ERR | Unable to load storage area for id 1: | zm_storage.cpp | 69
```
ZM still works though.

This is how the storage table looks like:
[image]

I used to have a second storage with the ID 1, but I deleted it. Maybe something doesn't get deleted in the DB when deleting storages and the deleted one is still referenced elsewhere?

closed 2019-01-06 by close-issue-app[bot]

connortechnology, 2019-01-06:
I suspect that the Monitor db entry is still referencing the deleted Storage Area.  Edit the monitor.

Warudo, 2019-01-06:
Thanks for respond even though I was a bad boy and ignored the template. :-)

Anyway, that was it. Both monitors showed "Default" (which is 0 and exists) for Storage in the UI, but the Storage ID was actually set to 1 in the Monitor table.

UI:
[image]
DB:
[image]

So you can't see this problem from within the UI.
Clicking "Save" again without actually changing anything fixed the problem.