ZoneMinder/zoneminder #2399: Some security vulnerabilities in v1.32.3

LoRexxar, 2019-01-08:
**Describe Your Environment**
- ZoneMinder v1.32.3

**Describe the bug**
some security vulnerabilities in v1.32.3 

**To Reproduce**
some details in [some datails](https://github.com/LoRexxar/CVE_Request/tree/master/zoneminder%20vul%20before%20v1.32.3)

**Expected behavior**
The above problems can lead to user information leakage, unauthorized operation, server command execution...

mnoorenberghe, 2019-01-21:
Progress:
- [ ] skins/classic/views/events.php line 44 sql injection
- [ ] sql query error Reflected xss
- [x] skins/classic/views/control.php line 35 second order sqli (#2422)
- [x] skins/classic/views/controlcap.php Reflected xss (#2423)
- [ ] includes/functions.php daemonControl command injection
- [x] ajax/status.php line 276 orderby sql injection (#2421)
- [ ] ajax/status.php line 393 sql injection

mnoorenberghe, 2019-01-21:
I think all issues are addressed once #2434 is merged. I've also been adding CSP for script-src to some pages to mitigate XSS in the future.

LoRexxar, 2019-01-22:
This sounds like a good decision.

closed 2019-02-09 by mnoorenberghe