ZoneMinder/zoneminder #2403: (FreeBSD) X264 encode doesn't work

abishai, 2019-01-09:
**Describe Your Environment**
1.32.3 under FreeBSD 11.2 (not a regression, 1.30 was broken in the same manner).

**If the issue concerns a camera**
For any camera (I tried jpeg and h264).

**Describe the bug**
X264 backend under FreeBSD doesn't work. It pretend to, but the resulting mp4 files have no tracks. .incomplete files have 160B size.

**To Reproduce**
Zoneminder setup under FreeBSD with X264 backend. I can provide test machine.

**Expected behavior**
Container with data.

**Debug Logs**
ZM doesn't print errors in logs and reports everything is fine.
Here is debug for mp4 container:
```
-rw-r--r--  1 www  www   633B Jan  9 10:10 /var/db/zoneminder/events/12/2019-01-09/2615/2615-video.mp4
```

```
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x80a651000] Could not find codec parameters for stream 0 (Video: h264 (avc1 / 0x31637661), none, 1280x1024): unspecified pixel format
Consider increasing the value for the 'analyzeduration' and 'probesize' options
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/var/db/zoneminder/events/12/2019-01-09/2615/2615-video.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42isom
    creation_time   : 2019-01-09T07:00:57.000000Z
  Duration: 00:00:00.00, bitrate: N/A
    Stream #0:0(und): Video: h264 (avc1 / 0x31637661), none, 1280x1024, 1k tbr, 1k tbn, 2k tbc (default)
    Metadata:
      creation_time   : 2019-01-09T07:00:57.000000Z
      encoder         : JVT/AVC Coding
```

knight-of-ni, 2019-01-09:
What is the cpu architecture?

abishai, 2019-01-09:
CPU: Intel(R) Xeon(R) CPU           X5680  @ 3.33GHz (3333.52-MHz K8-class CPU)
System is x64

abishai, 2019-01-09:
Also, reported here https://forums.zoneminder.com/viewtopic.php?f=38&t=27916&p=108926#p108926 under FreeNAS (FreeBSD based OS).

knight-of-ni, 2019-01-09:
Knowing the version of x264 and ffmepg might help too.

The latest development branch logs output from the ffmpeg api now. Can you run 1.33.x and provide a full page of output from the camera? I'm sure Isaac will need that to troubleshoot this.

knight-of-ni, 2019-01-09:
Or, if it is easier for you, you could patch #2321 into your build of 1.32.3:
https://patch-diff.githubusercontent.com/raw/ZoneMinder/zoneminder/pull/2321.patch

FYI, this is going to make your log very noisy.

abishai, 2019-01-09:
libx264-0.155.2917
ffmpeg version 4.0.2
But AFAIK, mp4v2 is used to handle X264 encoder. Actually, I suspect it. mp4v2 1.9.1

knight-of-ni, 2019-01-09:
I bumped the bounty on this by $150, but it appears you would not know it. Bounty Source no longer embeds an indicator into the issue since forever, and I have not been successful trying to fix it.

connortechnology, 2019-01-09:
yeah the x264 encode doesn't use ffmpeg libs, it uses mp4v2 and libx264.  I will eventually get around to spinning up my freebsd vm and doing some testing.

Long term I have a branch that does away with the x264 encode code and just uses ffmpeg.  However that will be a long time before we merge it.

abishai, 2019-01-10:
@knight-of-ni I've received a mail. Not sure if this issue worths $165 though :P
@connortechnology Do you need help with FreeBSD ?  You can get VM images here or I can provide you with a test machine. https://download.freebsd.org/ftp/releases/VM-IMAGES/12.0-RELEASE/amd64/Latest/

davidjb, 2019-03-11:
/unstale

arrowd, 2019-05-24:
I'd like to work on this, but need some concrete reproduction steps. I'm completely unfamiliar with ZoneMinder.

Is it possible to get reproduction without any camera?

connortechnology, 2019-05-24:
You can use a file as an input type.  This probably isn't a great issue for a first timer.

connortechnology, 2019-05-24:
Please note that since mp4v2 is deprecated in debian etc there isn't any point in trying to fix it.  We will be replacing the x264 encode code with ffmpeg based version.

arrowd, 2019-05-24:
So, there's nothing specific to FreeBSD, then? The bounty is set for implementing ffmpeg path?

abishai, 2019-05-25:
It's specific to FreeBSD. The problem is easily reproducible and sits in rather small amount of C code, however it can be tricky to solve.
Container files have no payload (but correct mp4 file structure) while function that copy data to container reports success.
The problem is that mp4v2 is well, abandonware and not many FreeBSD ports links with it. So, library itself can have issues on FreeBSD.
But maybe (just maybe) it cant be solved easily with just one LOC if you are lucky.

arrowd, 2019-05-25:
I tried looking into it but failed to set up things. I can look into that, if I'll be given some instructions on how to reproduce the problem.

abishai, 2019-05-25:
To say the truth, I never tried File source in FreeBSD, so probably you will be the first one, but...
After you installed ZM on FreeBSD, you should create monitor with File source type and point it to valid path in source path. On storage tab choose X264 encode and start monitor in Record mode.

abishai, 2019-05-25:
I was able to use ffmpeg backend for a file. Just point it to file:/path in source and set correct dimensions.

arrowd, 2019-06-02:
The fix for this bug has been committed both upstream and FreeBSD Ports tree. FreeBSD users should make sure they are using `zoneminder` package of version `1.32.3_2` or greater.

The bug can be closed, I guess.

Tsaukpaetra, 2019-06-02:
Sweet! Will await for it to trickle down...

closed 2019-06-03 by connortechnology