ZoneMinder/zoneminder #2406: Local tw68 camera problem

nickmcgill, 2019-01-09:
**Describe Your Environment**
-  ZoneMinder 1.32
- installed from PPA
- Ubuntu server 18.10

tw68,
Local camera

**Describe the bug**
Unable to access attached cameras

**To Reproduce**
Install zoneminder with a tw68 capture card

**Expected behavior**
Monitor attached cameras

**Debug Logs**
Currently unable to comply, as attempting to build zoneminder has broken the system (cmake insists on "mysqlclient") 
```

I think the problem is a logical error in line 955 of zm_local_camera.cpp which reads
    if ( (input.std != V4L2_STD_UNKNOWN) && vidioctl( vid_fd, VIDIOC_S_STD, &stdId ) < 0 )   {
rather than 
    if ( (input.std != V4L2_STD_UNKNOWN) && (vidioctl( vid_fd, VIDIOC_S_STD, &stdId ) < 0) )   {
which I think is should read.

connortechnology, 2019-01-09:
I have committed your fix. Let us know if it works.

nickmcgill, 2019-01-10:
Thanks for that. That gets me past that problem but I'm now running into 

zmc_dvideo0 | FAT | Failed to set video standard 255: Device or resource busy | zm_local_camera.cpp | 956
As far as I can tell, it should work - I don't know why it comes up as "busy".
I've attached the output from ls, zmu and v4l-info in case you have a chance to look at this.
I'm pretty sure that it's not a permissions thing, as I tried chmod 777 to no effect, and www-data is a member of video.

Thanks for your help

Nick

[video0-info.txt](https://github.com/ZoneMinder/zoneminder/files/2746104/video0-info.txt)

connortechnology, 2019-01-10:
I am going to need debug logs at level 9

connortechnology, 2019-01-10:
I suspect that the 255 is an invalid video standard. This code is not well documented, I'll have to do some reading. What have you selected for Palette under Monitor settings?

connortechnology, 2019-01-10:
My mistake, I mean format... please tell me all the settings on the source tab of the monitor.

nickmcgill, 2019-01-10:
Rookie question, How do I set the debug level to 9?

PAL appears to be 0xff

Source settings worked on older kernels with the tw68-v2 built (image attached)
I'm guessing that the newer tw68 may return something unexpected.

If it looks like taking up too much time, don't worry - I'll revert to an older kernel with the old tw68 module.

[image]

Thanks again

connortechnology, 2019-01-10:
It would be better to actually get it solved. Other people might have the same problem.  I made one small change, please test it.  To turn on debug logging, go to Options -> Logging.  Set LOG_LEVEL_FILE=>Debug, LOG_DEBUG=>checked LOG_DEBUG_LEVEL=>9, LOG_DEBUG_FILE=>empty , LOG_DEBUG_TARGET => _zmc  .  Then if your device is /dev/video0 the logging will go to /var/log/zm/zmc_dvideo0

connortechnology, 2019-01-10:
One thing more, you might want to choose 32bit instead of 24bit. it allows you to use faster functions.

nickmcgill, 2019-01-11:
The change you made gave this message
zmc_dvideo0 |   | 18106 | FAT | Failed to set video standard 255: 16 Device or resource busy | zm_local_camera.cpp | 957

Here's a log file - I hope it helps
[zm_debug.log.zip](https://github.com/ZoneMinder/zoneminder/files/2748620/zm_debug.log.zip)
I changed to 32-bit while the log file was being generated.

Thanks again

connortechnology, 2019-01-11:
So... one thought is that I don't think you can have 640x480 PAL.  You might try 720 × 576 Or try NTSC... Are you sure you want PAL?

Another thought is that apparently that error can happen if something is already capturing.  Do you have another monitor recording from that same channel?

connortechnology, 2019-01-11:
yeah it's not a permission problem, setting the format is the last command we send when setting it up.  We are definitely able to talk to the device.   I really think it's the resolution.  Would be really nice if when editing the monitor we could populate the width and height with a list of what the camera supports.  Doable.  Or may just add a check for PAL and invalid resolution.

nickmcgill, 2019-01-11:
@connortechnology I've had 640x480 PAL (and 720x680) working with tw68-v2 on old kernels.
New system, ubuntu server 18.10 and zoneminder + dev libs etc., nothing else accessing any video channels. I'm UK based, always used PAL...
@knight-of-ni 
"nick@zm-server:~$ sudo adduser www-data video
The user `www-data' is already a member of `video'."
@connortechnology Under the previous driver, that channel would display PAL 720x576, 640x480 and 320x240 with image distortion if the aspect ratio wasn't correct.

Thanks to you both

nickmcgill, 2019-01-11:
Change to 720x576 PAL failed, as did change to NTSC - both with "device busy"

(previous comment, "720x680" is a typo...)

knight-of-ni, 2019-01-11:
Can you view the device using a third party player, like vlc?
If you can, verify the resolution vlc reports matches what you entered into zoneminder.

nickmcgill, 2019-01-11:
I currently don't have a gui installed on the machine, so I haven't been able to try it.

I'll install one and see what I can find out.

nickmcgill, 2019-01-14:
It appears that there's a problem with the tw68 module - vlc crashes when I select any of the /dev/video devices.

VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
Segmentation fault (core dumped)
I've attached debug log files from vlc, but my feeling is that this is beyond the scope of zoneminder.

Thanks for your help so far - it's back to the old kernel for me I think.
[vlc-logs.zip](https://github.com/ZoneMinder/zoneminder/files/2755937/vlc-logs.zip)

closed 2019-03-22 by stale[bot]

TP234-CPE, 2021-01-02:
Hello, I waking up this closed topic because I have the same issue with a tw68 card. I tried a runaround by setting the source type to libvlc but it can't run on different channel and I have multiple camera on the same device. So i went back to the source type local and I can't get it to work. Same "device or resource busy" issue. 

Here is the log I get :
`2021-01-02 15:56:48	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 15:56:48	zmc_dvideo0		6658	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 15:56:48	zmc_dvideo0		6658	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 15:56:38	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 15:56:38	zmc_dvideo0		6656	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 15:56:38	zmc_dvideo0		6656	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 15:56:36	zmc_dvideo0		6655	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 15:56:36	zmc_dvideo0		6655	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 15:56:33	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 15:56:33	zmc_dvideo0		6653	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 15:56:33	zmc_dvideo0		6653	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 15:56:32	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 15:56:31	zmc_dvideo0		6652	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 15:56:31	zmc_dvideo0		6652	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926`

Did someone found a solution/can help me with that ? What more information would you need ?

TP234-CPE, 2021-01-02:
Hello, I waking up this closed topic because I have the same issue with a tw68 card. I tried a runaround by setting the source type to libvlc but it can't run on different channel and I have multiple camera on the same device. So i went back to the source type local and I can't get it to work. Same "device or resource busy" issue.

Here is the log I get :
```
2021-01-02 16:07:58	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 16:07:38	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 16:07:38	zmc_dvideo0		6761	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 16:07:38	zmc_dvideo0		6761	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 16:07:28	zmc_dvideo0		6760	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 16:07:28	zmc_dvideo0		6760	FAT	Failed to set video standard 255: 16 Device or resource busy
```

Did someone found a solution/can help me with that ? What more information would you need ?