ZoneMinder/zoneminder #2408: Unable to set non-US date format for Event naming

hunnymonster, 2019-01-11:
**THIS FORUM IS FOR BUG REPORTS ONLY**

Do not post feature or enhancement requests, general discussions, or support questions here.

Feature and enhancement requests, general discussions, and support questions should occur in one of the following areas:

- The [ZoneMinder-Chat Slack channel](https://zoneminder-chat.herokuapp.com/)
- The [ZoneMinder Forum](https://forums.zoneminder.com/)

Docker related issues should be posted here: https://github.com/ZoneMinder/zmdockerfiles

In order to submit a bug report, please populate the fields below this line. This is required.

----------------------------------------------------------------------------------------------------

**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] : 1.32.3
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] : PPA
- Full name and version of OS: Ubuntu 18.04.1 (kernel 4.15.0-43)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
With all date/time configs set to display %d/%m/%Y %H:%M:%S the view of Events (at localhost/zm/index.php?view=events&page=1...) continues to display in US format (ie. %m/%d)
Noted (by others) in forum - https://forums.zoneminder.com/viewtopic.php?f=38&t=27534&p=108583

**To Reproduce**
Steps to reproduce the behavior:
1. Set locale to en_gb under Options > System > LANG_DEFAULT
2. Set timezone to Europe/London in both /etc/php/x.x/cli/php.ini & /etc/php/x.x/apache2/php.ini
3. service stop zoneminder
4. service stop apache2
5. service start apache2
6. service start zoneminder
7. generate event
8. view events page

**Expected behavior**
Date format in event view should be consistent with the date format elsewhere.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

closed 2019-03-19 by stale[bot]

spacelama, 2021-09-26:
That stalebot is overly aggressive.  No bug is going to be fixed in 2 months, regardless of whether there's been comments or not.

This bug most definitely still exists in 1.34.26.  It's also annoying because apart from the date format of events being nonsensical for 95% of the world, it also doesn't include year, so it's not obvious to see when zoneminder is hanging onto year old events needlessly.   A mm/dd or dd/mm date field is *inherently* ambiguous, and can only be resolved when the day is between 13 and 31.

SteveGilvarry, 2021-09-26:
I would say 2.5 years between comments makes it pretty stale, and if no one in the very small community that actually fix bugs decides to work on a bug inside 2 months it is unlikely to get much air time. And then they can label it to prevent closure. Of course people can always reengage or log a new one to see if someone wants to look at it for them. And hence here we are.
For the year to be honest I had never even noticed as I don't keep recordings long enough to need a year, and even month/day my brain must have been able to work it out without issue.
Anyway language settings already has the ability to fix it, so I added them to en_gb, assuming you are running that language, since we don't have en_AU.
Details here if you want to edit your file locally until the fix cascades through, you can format as you like then using php format options. Adding year may impact table size so will await next bug report about date being too long.
https://github.com/ZoneMinder/zoneminder/pull/3360/files

spacelama, 2021-10-03:
That looks good to me, thanks.

My issue with stalebots is the interaction with github - only the original submitter can reopen an issue, so what's left is to submit a duplicate issue, or comment on an old issue and hope someone notices the comment - apparently not terribly successful on most github projects.  Thanks for noticing and even more thanks for fixing!

commit f54e779b (2021-09-27): Merge pull request #3360 from ZoneMinder/strftime_en_gb_format

Define Date formats in lang file for en_gb resolves #2408