ZoneMinder/zoneminder #2412: Events not showing under view when not using /zm folder - v1.32.3

MalakymR, 2019-01-15:
**Describe Your Environment**
- 1.32.3
- Source/Release
- Gentoo

**Describe the bug**
Events do not show under monitor view when not using a /zm/ sub-folder in apache. This seems to be due to     'PathToIndex' => '/zm/index.php', in inclues/Server.php

Is there a config to override this variable in the .conf files?

**Expected behavior**
Show events under view=watch

**Debug Logs**
```
[Tue Jan 15 14:49:39.838264 2019] [:error] [pid 27771:tid 139725402724096] [client 10.0.0.80:57930] script '/usr/share/zoneminder/www/zm/index.php' not found or unable to stat, referer: http://server/index.php?view=watch&mid=9

<VirtualHost *:80>
  DocumentRoot /usr/share/zoneminder/www
  *etc
</VirtualHost>

```

Nocifer, 2019-01-19:
Looks like another case of the issue outlined in #2253, but the fix provided in #2377 is not yet included in an official release. You could try building the latest master or, since this is non-compiled php and js we're talking about, you could edit your existing ZM files and apply the fix yourself. In either case, your issue should be resolved.

joshuacox, 2019-01-22:
I get lot's of these errors in my browser console:

```
TypeError: server.PathToIndex is not a function
```

adding for searchability

MalakymR, 2019-01-29:
@Nocifer We know we could fix this issue - and have a second box that runs closer to master. Can't keep running bleeding edge all of the time.

We've kind of given up shoehorning ZM back to / and just use /zm/ with a redirect from the / folder now - sub folders should not be the default imo - anyone that wants a sub folder can do that easily.

If this will be fixed by the next release I can wait for that.

joshuacox, 2019-01-29:
hmm, I'm noticing these errors while using the subdir `/zm'  should I be opening a separate issue?

joshuacox, 2019-03-30:
/bump this issue is still happening for me

closed 2019-06-05 by stale[bot]

JasonSanDiego, 2020-07-01:
I'm having this issue as well, running 1.34.16. When in montage review, the images don't load (just get No Data in place of live images) and I see this in the browser console:
index.php:1 GET https://example.com/zm/index.php?view=image&eid=249190&fid=48&width=336&height=252 403 (Forbidden)

I have zoneminder running under a different path from /zm, and everything else works, as far as I can tell. Just montage review is broken.