ZoneMinder/zoneminder #2435: syntax error at /usr/share/perl5/ZoneMinder/Filter.pm

zhuykovkb, 2019-01-22:
**Describe Your Environment**
ZM version from commit 2914fb1d58c9afc14e376c90f523ba7555bdc125
Build zm from dockerfiles

Log button is red. Tapped on it i see:
[image]

Run from console:

```
root@8abd3312a438:/ZoneMinder# zmfilter.pl --filter_id=6 --daemon
Bareword found where operator expected at /usr/share/perl5/ZoneMinder/Filter.pm line 151, near "// See"
        (Missing operator before See?)
syntax error at /usr/share/perl5/ZoneMinder/Filter.pm line 151, near "// See getFilterQueryConjunctionTypes"
Global symbol "$term" requires explicit package name (did you forget to declare "my $term"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 152.
Global symbol "$term" requires explicit package name (did you forget to declare "my $term"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 153.
Global symbol "$term" requires explicit package name (did you forget to declare "my $term"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 155.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 156.
Global symbol "$term" requires explicit package name (did you forget to declare "my $term"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 156.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 163.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 165.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 167.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 169.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 172.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 174.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 176.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 178.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 180.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 182.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 186.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 188.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 190.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 192.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 196.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 197.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 199.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 200.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 202.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 203.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 205.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 206.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 208.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 220.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 226.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 230.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 243.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 255.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 267.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 280.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 282.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 285.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 287.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 289.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 292.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 294.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 296.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 298.
Global symbol "$self" requires explicit package name (did you forget to declare "my $self"?) at /usr/share/perl5/ZoneMinder/Filter.pm line 302.
syntax error at /usr/share/perl5/ZoneMinder/Filter.pm line 304, near "}"
/usr/share/perl5/ZoneMinder/Filter.pm has too many errors.
Compilation failed in require at /usr/local/bin/zmfilter.pl line 93.
root@8abd3312a438:/ZoneMinder# 

```

closed 2019-01-22 by close-issue-app[bot]

connortechnology, 2019-01-22:
Fixed, new packages building

zhuykovkb, 2019-01-22:
u are awesome! thank you, it solved the problem :+1: