ZoneMinder/zoneminder #2436: Reflective XSS vulnerability

360CodeSafe, 2019-01-23:
**Describe Your Environment**
-ZoneMinder v1.32.3

**Describe the bug**
Reflective XSS vulnerability

**To Reproduce**
ZoneMinder has one reflective XSS vulnerabilities
1.plugin.php  line 106
The parameters in the request are accepted at 39 lines, and the page is output at 106 lines without verification.
[image]
[image]

**Expected behavior**
The vulnerability mcan lead to user information leakage, unauthorized operation

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

mnoorenberghe, 2019-01-24:
I can confirm the issue: 
```/zm/index.php?view=plugin&pl=%3Cimg%20src=%22%22%20onerror=%22alert(%27foo%27)%22%20/%3E&mid=1&zid=1```

closed 2019-01-24 by knight-of-ni

carnil, 2019-01-24:
CVE-2019-6777 was assigned for this issue.

commit 59cc6541 (2019-01-24): plugin.php: Fix XSS and directory traversal bugs. Fixes #2436

This view seems like dead code so maybe it should be removed instead.