ZoneMinder/zoneminder #2470: Weak Cross-site Resource Forgery(CSRF) Protection (1)

Loginsoft-Research, 2019-01-24:
**Describe Your Environment**
- ZoneMinder v1.33.1
- Installed from - ppa:iconnor/zoneminder-master 

**Describe the bug**
Each form generates & adds a unique CSRF tokens(__csrf_magic) to every form on the webpage

Expected behavior - Every token should be unique to it's own form& shouldn't be valid in other form.

Current scenario - Tokens of Form A,can be used in Form B.
(If tokens aren't expired - Expiry time, Session destroyed)

**To Reproduce**
  
1. Open Form A -
Example- http://localhost/zm/?view=filter&sort_field=StartTime&sort_asc=1
(key:7fa4db1be2269f5a488147d53f405ee07d56a0f8,1544874804)

2. Open Form B -
Example URL- http://localhost/zm/index.php

POST Data -
`__csrf_magic= key:32d4a3818dccbc2a05b59247358bd8498a2729a9,1544875131 &view=group&action=group&gid=&newGroup%5BName%5D=test&newGroup%5BParentId%5D=1&saveBtn=Save:`

3. Use the tokens of Form A, in Form B request & forward it.

4. Request will be successfully initiated.

**Expected behavior**
Session should be unique to each form.

**Debug Logs**
```

None

```

mnoorenberghe, 2019-02-10:
While I understand the value of not using the same CSRF token, it's likely to break things to retrofit this into ZM. I think it wouldn't be too hard to take the `view` into account when calculating the CSRF token but I'm not confident that the same `view` parameter is always sent back with POST requests which would mean the user would get an error. I also wouldn't be surprised if there are some forms that are loaded in one `view` that submit with a different `view` argument which also wouldn't work.
I think this is something we can try in the future, potentially with a front-end re-work but in the short term I think #2507 is much more important: any CSRF protection is better than none.

Sjord, 2019-02-27:
@Loginsoft-Research What is the problem with one CSRF token across all forms? Do you perhaps have an attack scenario?

Loginsoft-Research, 2019-02-28:
Expected behavior - Every token should be unique to it's own form & shouldn't be valid in other form.

Current scenario - Tokens of Form A, can be used in Form B.

An attacker can steal a CSRF token via XSS attack & reuse it on other forms.

closed 2019-05-06 by stale[bot]