ZoneMinder/zoneminder #2471: Improper Session Management - Session Fixation

Loginsoft-Research, 2019-01-24:
**Describe Your Environment**
- ZoneMinder v1.33.1
- Installed from - ppa:iconnor/zoneminder-master 

**Describe the bug**
Before any response being rendered on the web page, a cookie is being set as part of response via Set-Cookie header.Currently due to improper implementation, multiple cookies (3-5) are being set when a user successfully logs in, which isn't an expected behavior.Now when the next user logs into the application using the same browser, the last user's last cookie is being set as the present user's first cookie.

**To Reproduce**

- Login into the application, capture the response (Live HTTP header addon for firefox)

- Multiple Set-Cookie headers will be present in the response
  

Exploitation

Actors-

- User A (Victim)
- User B (Attacker)

1. User B logs into the application, captures his own Cookies & logs out.

User B's cookie :
Set-Cookie: ZMSESSID=rrnilufc9vgb3cp0l2m7cqrc91; path=/; HttpOnly
Set-Cookie: ZMSESSID=blkta1mgocj5ksqdg5ncpdptg3; path=/; HttpOnly
Set-Cookie: ZMSESSID=blkta1mgocj5ksqdg5ncpdptg3; path=/; HttpOnly
Set-Cookie: ZMSESSID=blkta1mgocj5ksqdg5ncpdptg3; path=/; HttpOnly
Set-Cookie: ZMSESSID=p79d4mk2g6sm5qi6o51ep6j6m5; path=/; HttpOnly - `Common to User A's First cookie`

2. User A Uses the same browser & log into the application.

User A's cookie -
Set-Cookie: ZMSESSID=p79d4mk2g6sm5qi6o51ep6j6m5; path=/; HttpOnly - `Common to User B's Last cookie`

Set-Cookie: ZMSESSID=2397j5pchtgt153ukrmutgbmv1; path=/; HttpOnly
Set-Cookie: ZMSESSID=2397j5pchtgt153ukrmutgbmv1; path=/; HttpOnly
Set-Cookie: ZMSESSID=2397j5pchtgt153ukrmutgbmv1; path=/; HttpOnly
Set-Cookie: ZMSESSID=eg5hvsn3i67n34fibt5nq7lbu6; path=/; HttpOnly

3. User B uses User A's Last cookie to hijack the User B's session (locally/remotely)

**Expected behavior**
- Only set single cookie the user & invalidate the same once user logs out of the application.

**Debug Logs**
```

None

```

galaktipus, 2021-03-01:
the issue was assigned CVE-2019-7350. Any plans to commit a fix?

synthead, 2023-12-19:
When is this CVE planned to be resolved?