ZoneMinder/zoneminder #2477: Failure to invalidate session

Loginsoft-Research, 2019-01-25:
**Describe Your Environment**
- ZoneMinder v1.33.1
- Installed from - ppa:iconnor/zoneminder-master 

**Describe the bug**
When user logged-in and stays idle, the session should be destroyed after a certain period of time. Also the reusability of old session is possible, even after we close the application tabs in browser, which can cause an unauthorized access to the application.

**To Reproduce**
  Affected URL :
http://localhost/zm/index.php

- Log into Zoneminder as any user. 
- Leave the session idle for longer period, the session never gets invalidated.

An attacker can take advent of the weakness and gain unauthorized access to the web application without providing any credentials.

[image]

**Expected behavior**
 - Always specify session expiration date, make sure it is not to long and that the user can reset the session using the log off functionality of the application 

**Debug Logs**
```

None

```

mnoorenberghe, 2019-02-10:
This is normal for modern web-apps e.g. Facebook keeps you logged in for months at a time. The administrator can control the session lifetime via PHP configuration and users can manually log out from our UI so I would consider this report invalid personally. I'll let others chime in.

Sjord, 2019-02-28:
> An attacker can take advent of the weakness and gain unauthorized access to the web application without providing any credentials.

I don't think this is as straightforward as this sentence implies. Even if sessions last forever it is pretty hard to obtain a valid session token.

I would find a session timeout of several days acceptable. I think PHP's gc_maxlifetime makes sure that sessions don't last forever, but I am not sure on the details of how that works.

mnoorenberghe, 2019-03-11:
This was fixed by session length UI in #2487

closed 2019-03-11 by mnoorenberghe