ZoneMinder/zoneminder #2493: Bandwidth change not working

zd59, 2019-02-01:
**Describe Your Environment**
-ZoneMinder GIT: https://github.com/ZoneMinder/ZoneMinder.git from 31.1.2019 13.46 GMT+1 with commit Isaac Connor "fix state changing/etc"

ZoneMinder installed from a GIT source, compiled & packed with adapted zoneminder.SlackBuild form slackbuilds.org WEB page

- Full name and version of OS
Slackware64-current, updated 31.1.2019, kernel 4.19.16 
**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
GUI click on "Bandwidth" button it happens one of below, by pure chance:
open a new window, & select other than current one closes the window, but there is no change
or  the complete current window is replaced with selecting a bandwidth, when selecting & click on "save", the window wipes and become white without a single character on it.
Than you have to restart GUI - open a startup page again

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'bandwidth'
2. Click on 'current bandwidth'
3. Scroll down to 'other than current'
4. Click on "save" button
4. See error

**Expected behavior**
"save" button changes active bandwidth.

**Debug Logs**
```

 zm_debug.log.29074
```
02/01/19 12:07:14.833059 web_php[29074].DBG [View: options Request:  Action: ] at /var/www/htdocs/zm/index.php line 197
02/01/19 12:07:30.191570 web_php[29074].DBG [View: request Request: status Action: ] at /var/www/htdocs/zm/index.php line 197
02/01/19 12:07:32.903269 web_php[29074].DBG [View: bandwidth Request:  Action: ] at /var/www/htdocs/zm/index.php line 197
02/01/19 12:07:42.387154 web_php[29074].DBG [View: none Request:  Action: bandwidth] at /var/www/htdocs/zm/index.php line 197
02/01/19 12:07:42.387200 web_php[29074].WAR [No includes/actions/none.php for action bandwidth] at /var/www/htdocs/zm/index.php line 218

mnoorenberghe, 2019-02-10:
I can reproduce this bug… it's probably due to the inline event handler changes.

mnoorenberghe, 2019-02-10:
Now I'm thinking it was from dbe9817bc823225e3d00ea880fa0001c925b3d15.

closed 2019-02-10 by mnoorenberghe

zd59, 2019-02-11:
Confirmed working.
Congratulate, fastest response and a solution.

PR #2511 (2019-02-10): bandwidth.php: Submit to the 'bandwidth' view but render the 'none' view
Fixes #2493
commit cdbd59f0 (2019-02-10): bandwidth.php: Submit to the 'bandwidth' view but render the 'none' view. Fixes #2493