ZoneMinder/zoneminder #2495: Update Brazilian Portuguese translation

mgsantana, 2019-02-01:
**Describe Your Environment**
- ZoneMinder 1.32.3
- Installed from Debian repository.
- Debian GNU/Linux 9.7

**Describe the bug**
- The Brazilian Portuguese translation is out-dated.

**Expected behavior**
- To update the Brazilian Portuguese translation of ZoneMinder.

SteveGilvarry, 2019-02-02:
Are you able to assist? All the translations are community supplied, so if you able to update it would help yourself and others.
You can edit it directly in GitHub
https://github.com/ZoneMinder/zoneminder/blob/master/web/lang/pt_br.php

mgsantana, 2019-02-04:
Hello Steve,
    I'm already working on it! ;-)
    Regards,
    -- 
Marcelo Santana (aka msantana) <marcelo@msantana.eng.br>
4096R/89C55467: 88FB 5D63 ED02 3B5D 90D6  3A3E 8698 1CC9 89C5 5467

    On 2/2/19 2:09 AM, Steve Gilvarry
      wrote:
    
    
      
      Are you able to assist? All the translations are community
        supplied, so if you able to update it would help yourself and
        others.
        You can edit it directly in GitHub
        https://github.com/ZoneMinder/zoneminder/blob/master/web/lang/pt_br.php
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ZoneMinder/zoneminder/issues/2495#issuecomment-459936326",
"url": "https://github.com/ZoneMinder/zoneminder/issues/2495#issuecomment-459936326",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

mnoorenberghe, 2019-02-10:
Looking forward to the pull request but I'll close this for now since I don't think we need to track this as an issue.

closed 2019-02-10 by mnoorenberghe