ZoneMinder/zoneminder #2498: Custom filters don't show up

phillijw, 2019-02-06:
**Describe Your Environment**
- v1.33.1
- docker container of quantumobject/docker-zoneminder

**Describe the bug**
After adding a new filter, the filter does not actually get added. I can watch network tab and see that the page posts and receives a 200 OK back but the filter never shows up in the dropdown list. The filter I add is:
* Name: DeleteOld
* `End Date/Time` `less than or equal to` `-1 day`
* Sort by `id` `Asc` Limit first `99999` results only
* Delete all matches
* Then I hit Save or Save As and nothing happens

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Filter page
2. Enter the filter as above
3. Hit save

**Expected behavior**
I expect a new filter to be created and shown in the drop down list. This does not take place.

**Debug Logs**

Date/Time | Component | Server | PID | Level | Message | File | Line
-- | -- | -- | -- | -- | -- | -- | --
2019-02-05 22:06:36 | web_php |   | 12821 | WAR | No includes/actions/filter.php for action SaveAs | /usr/share/zoneminder/www/index.php | 218
2019-02-05 22:04:02 | web_php |   | 3779 | WAR | No includes/actions/filter.php for action Save | /usr/share/zoneminder/www/index.php | 218

mnoorenberghe, 2019-02-10:
Did this work for you in the past?

phillijw, 2019-02-10:
Yeah, I just recently upgraded to the newest docker image and I can no longer create filters. This was on a fresh install (didn't reuse old database or config or anything).

connortechnology, 2019-02-20:
I think I have fixed this.  Please confirm and close if it is fixed.

phillijw, 2019-02-20:
@connortechnology Would you mind elaborating a little bit? Which version is the fix in? Which commit? Etc... I use the docker images and I'm not sure they're updated yet

phillijw, 2019-02-21:
This is the image I use: quantumobject/docker-zoneminder:latest

Is this the wrong one? The latest one I just downloaded is 1.32.3 but my existing zm installation is 1.33.1. I REALLY wish whoever created those images would tag them with actual versions and not rely on "latest".

TorbjornZachrisson, 2019-03-07:
Why is the latest version of the container an older version of zoneminder than the one I pulled a few weeks ago. Very annoying when the database is upgraded to an later version... The old container was 1.33.1.

phillijw, 2019-03-09:
Here's the problem/solution that I had. I guess it doesn't relate directly to this repo (which I didn't realize at first).

https://github.com/QuantumObject/docker-zoneminder/issues/67

closed 2019-03-09 by phillijw