ZoneMinder/zoneminder #2499: Camera event buttons not working

tdragonite, 2019-02-06:
**Describe Your Environment**
- 1.33.1~20190205172123-bionic
- PPA Master Snapshot
- Ubuntu 18.10

**Describe the bug**
The camera event buttons aren't working, Stop/Play still work. Next/Rewind/Step Forward/Fast Forward/Prev don't work. The buttons still look enabled and the tooltips pop up.

**To Reproduce**
Steps to reproduce the behavior:
1. Open one event
2. Click on Next/Prev/Rewind/Step Forward/Fast Forward

**Expected behavior**
Click on Next for the next event, Prev for prev event, ecc.

mnoorenberghe, 2019-02-10:
Hello, what browser are you using? Do you know roughly when this started happening?

SteveGilvarry, 2019-02-10:
https://github.com/ZoneMinder/zoneminder/commit/10391498669d20b3c79933c66fe5f7687b8d9f99
I think this was fixed

mnoorenberghe, 2019-02-10:
@tdragonite Can you confirm this is fixed in the latest master builds?

tdragonite, 2019-02-11:
@mnoorenberghe yes, fixed in the latest master build :)

closed 2019-02-11 by SteveGilvarry