ZoneMinder/zoneminder #2500: MP4 passthrough produces corrupt videos.

poldon, 2019-02-06:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.1~20190122105611-bionic
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04.1 LTS

**Describe the bug**
Most of the time, the MP4 files have either choppy audio, are cut off (e.g. only the first 2 seconds of the event are saved), or completely corrupted (just a big X in the video for the event).  I tried setting the pre-event frames to 0 like someone else suggested, but it didn't make a difference.  I have 5 cameras, all of which worked fine before (1.32.x master PPA), but they all exhibit the same behavior.  I thought I'd rollback, but I don't think the database would work at that point.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on MP4 passthrough on the monitor(s).
2. Observe corrupt MP4 event files.

**Expected behavior**
Proper, non-corrupt videos.

**Debug Logs**
Will provide later if needed.  I don't see anything relevant so far.

poldon, 2019-02-20:
I figured out what is happening, but not the exact cause yet (I haven't looked at the code).  The videos have audio as well, and the timestamps for the audio must be incorrect.  So, what happens is that you get choppy audio because each audio packet is being spaced out further apart than it should be.  The video looks correct and at the right speed, but the audio continues for about a minute after the video has stopped if the event is a few seconds long.

My guess is that the audio PTS values are incorrect or something similar to that.

connortechnology, 2019-02-20:
Your build is a bit old.  I've done some work recently to improve the audio.  Please update and give it a try.

poldon, 2019-02-20:
Looks like it's working now!  Thanks!

closed 2019-02-20 by poldon