ZoneMinder/zoneminder #2502: System Timezone of Asia - Hong Kong will stop Montage Review from working normally (re-post)

dtongcc, 2019-02-07:
**Describe Your Environment**
Operating system:
Ubuntu Server 18.10 with updates from apt upgrade command.

Version of ZoneMinder tested (I followed [this guide](https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html#easy-way-ubuntu-16-04):
1.32 (installed from ppa:iconnor/zoneminder-1.32)
1.33.1 (installed from source)

**Describe the bug**
I have encountered a strange behavior in the Montage Review session that if the system timezone (not the timezone configured in php.ini) is Asia -> Hong Kong. The Montage Review will not be able to display past recordings.

**To Reproduce**
1. Fresh install with official Ubuntu Server image, set the timezone to Asia - Macau during the installation and follow the guide to install ZoneMinder. The timezone in the php.ini is set to Asia/Hong_Kong. Note that Macau and Hong Kong are both GMT+8:00.
$ sudo dpkg-reconfigure tzdata

2. Start the ZoneMinder and add monitors in Record mode, notice that the Montage Review session works normally.

3. On another system, follow the exact same steps other than setting the timezone to Asia - Hong Kong during Ubuntu Server installation. (Note again that the timezone in the php.ini is still set to Asia/Hong_Kong)

4. Start the ZoneMinder and add the exact same monitors in Record mode. The Montage Review will display "no data" for all previous recordings.

**Expected behavior**
Montage Review should work.

**Debug Logs**
There is no warning messages in Log console as well as apache log file. I can re-produce the error in VMWare Workstation, Ubuntu KVM guests and a Thinkpad notebook. Request for additional logs are welcome as I did not delete the VMs.

connortechnology, 2019-04-08:
Since mysql uses the system timezone, ZM will get very confused if system time zone and php time zone are different.

dtongcc, 2019-04-08:
I am sorry for my poor English, but what I mean is that when setting both system timezone and php.ini to Hong Kong will lead to mis-behavior of Montage Review. The timezone of "Macau" is my workaround but setting it back to Hong Kong will yield error again.