ZoneMinder/zoneminder #2503: Need "Storage Area Percent" Filter (for PurgeWhenFull on a Storage Area)

brwainer, 2019-02-08:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: installed from RPMFusion following exact instructions
- Full name and version of OS: CentOS Linux release 7.6.1810 (Core) - Kernel 3.10.0-957.5.1.el7.x86_64

**Describe the bug**
On a system where all footage is stored in a storage area that is not on the main disk, PurgeWhenFull will never return any results because "Disk Percent" never increases (or at least not due to recordings). My suggested fix is to introduce a new filter for "Storage Area Percent" which should match the percentage of the storage area for each individual recording. Users can combine this with the "Storage Area" Filter if they only want to affect a single filter. I am not requesting a change to the default PurgeWhenFull filter, although that is something I would ask to be considered as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure new server as desired.
2. Attach and mount a second disk. In my case it is LVM plus XFS, mounted on /mnt/recordings
3. Create a Storage Area on the new mountpoint. In my case it is named "Main" and is directed at /mnt/recordings/events
4. Configure monitor(s) to record to this storage area. No recordings should go to any location other than Main.
5. Once sufficient recordings have been made, notice that "PurgeWhenFull" is not deleting any recordings. At the top of my console I have "Storage: Main: 98%, Default: 10% /dev/shm: 19%".
6. Test the PurgeWhenFull filter and notice that no events are listed unless "Disk Percentage" is reduced to 10% or lower (matching whatever percentage is show for "Default").

**Expected behavior**
I understand the default behavior of PurgeWhenFull is only looking at the main disk percent, of the disk (partition) that is mounted as "/". Once a new filter is added, or some other resolution is made, I expect to be able to auto-delete the oldest recordings when a Storage Area that is not on the main disk fills up. This does not need to be the default behavior of PurgeWhenFull, but I expect to only have to change "Disk Percent" to "Storage Area Percent".

**Debug Logs**
No logs to attach at this time, please let me know if there is anything that would be helpful to you.

connortechnology, 2019-02-08:
It does when you specify the Storage Area in the filter.  Not well documented I know, 

So just add a Storage Area = to your filter and it will operate on that storage area.

brwainer, 2019-02-08:
Thank you that resolves the issue, but was not intuitive. I’m closing this as resolved.

closed 2019-02-08 by brwainer