ZoneMinder/zoneminder #2506: PurgeWhenFUll filter not run if Id is not 1

brwainer, 2019-02-09:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: installed from RPMFusion following exact instructions
- Full name and version of OS: CentOS Linux release 7.6.1810 (Core) - Kernel 3.10.0-957.5.1.el7.x86_64

**Describe the bug**
If the user makes a change to PurgeWhenFull and "Saves As" then deletes the original filter, instead of using "Save", the new PurgeWhenFull filter will have an Id that is not 1 (in my case it was 7). This causes an error from zmfilter.pl every 6 minutes from https://github.com/ZoneMinder/zoneminder/blob/2dc935b488e2935add61bfafefd88230a41b9851/scripts/zmfilter.pl.in#L262 Warning("No filter found for $sql with values(@sql_values)"); The quickest fix for me was to log into the database and change the Id to 1, and which point the filter worked properly again. 

Possible Causes (I have not done a full review of the source code):
-More than one filter with the same name, even after one is deleted? (How does the system know which Id to replace into the "WHERE Id = ?" on https://github.com/ZoneMinder/zoneminder/blob/2dc935b488e2935add61bfafefd88230a41b9851/scripts/zmfilter.pl.in#L229 when the SQL is prepared?)
-PurgeWhenFull hardcoded to an Id of 1?

It is also possible that this bug would affect any filter that is auto-executed that goes through the steps of "Save As" followed by deleting the original, I do not know (and have not tested) whether this is isolated to PurgeWhenFull.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the default PurgeWhenFull filter
2. Make a change, e.g. add a Storage Area criteria
3. Save As.
4. Delete the "Id 1 PurgeWhenFull" filter
5. Review the logs after 6 minutes (or FILTER_RELOAD_DELAY) and see the SQL error
6. directly in the database, change the filter's Id back to 1, via: "UPDATE Filters SET Id = 1 WHERE Id = 7"

**Expected behavior**
Using Save As on a filter followed by deleting the original should not cause the filter reload process to subsequently fail.

**Debug Logs**
No logs to attach at this time, please let me know if there is anything that would be helpful to you.

connortechnology, 2019-02-10:
The issue here is that the background zm_filter process are not being notified that a change has been made.  There used to be 1 single zm_filter.pl process, not there is one for every background filter. So when a filter is saved, we need to call zmdc.pl to restart/stop/stop whatever is appropriate.

knight-of-ni, 2019-02-12:
fixed via #2518

closed 2019-02-12 by knight-of-ni