ZoneMinder/zoneminder #2510: No picture after upgrade from 1.30 to 1.32 on ubuntu 16.04

ebeng, 2019-02-10:
all,

very simple; i've upgraded through (aptget update/upgrade/dist-upgrade) Zoneminder to 1.32.
Seems to be a typo in the following file:

changed in file;

/etc/apache2/conf-enabled/zoneminder.conf LINE 26

```
<Directory /usr/share/zoneminder/www>
to
<Directory "/usr/share/zoneminder/www">
```

After this, restart Zoneminder and it started to work back again.

connortechnology, 2019-02-10:
Those quotes should not be necessary.  I'm happy that adding them worked for you, but I suspect the issue was elsewhere.

closed 2019-02-10 by connortechnology

ebeng, 2019-02-10:
Well how could the issue be somewhere else?
Just please try to understand that, ive restarted zoneminder and reboted the vm a couple of times without any luck seeing the images. 

From the logs, only what was the error:

Feb 10 15:23:42 zoneminder web_php[1351]: ERR [getStreamCmdResponse stream error: Socket /var/run/zm/zms-342639s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()]

Exactly after putting the quotes around the directory on that line it started to work. 

I just reverted back again without quotes, it stopped showing images again after restaring zoneminder...