ZoneMinder/zoneminder #2516: mp4v2 is obsolete

onlyjob, 2019-02-11:
mp4v2 [has been removed from Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921127):

> mp4v2 is dead upstream and better alternatives exist.

SteveGilvarry, 2019-02-13:
@connortechnology What do you think? Swap this over to ffmpeg?

mastertheknife, 2019-02-18:
I don't think we should remove the X264:MP4V2 encoder. Me and others are using it, so far without issues.
The VideoWriter class is an abstract class, it was designed that way so more video writers can be added over time.
In the past i also planned to add x265 because i abandoned it because lack of time, and high encoding CPU usage.

If anything, i think we should implement ffmpeg:ffmpeg video writer. What i mean is use ffmpeg for encoding (take advantage of hw acceleration) and also ffmpeg for muxing the frames into a mp4 file.

jwtoler, 2019-06-08:
How can we install mp4v2 in Debian Stretch? I have installed everything related to a google search for mp4v2. I keep getting "fatal error mp4.h no such file or directory" error when trying to build from source using the docker method.

connortechnology, 2019-06-10:
You need libmp4v2-dev

sscraggles, 2019-06-19:
getting "fatal error mp4.h no such file or directory" when using the docker build of latest commit.  I see the dsc file is installing libmp4v2-dev.

EtoTen, 2019-06-30:
This affects Ubuntu 19.04 Disco, as mp4v2 is not available via the repositories.

kc2zgu, 2020-07-17:
Switching this to ffmpeg would allow using other containers, which could potentially lift the restriction of not being able to view video events as they're being recorded (MP4 is somewhat unique among containers in that it doesn't allow live streaming)

connortechnology, 2021-02-03:
Since 1.35.17 we have removed the mp4v2 code and depenency.

closed 2021-02-03 by connortechnology