ZoneMinder/zoneminder #2527: Floureon preset not working

koetj, 2019-02-16:
Hi,

Im running zm 1.32 and checked that the floureon ptz preset is not working.

I checked and the command for calling preset should be:

` my $cmd = "/form/presetSet?flag=4&existFlag=1&language=cn&presetNum=".$preset;`

For storing presets it should be

` my $cmd = "/form/presetSet?flag=3&existFlag=1&language=cn&presetNum=".$preset;`

Only storing is not working, don't know why but it can be done by the camera's webif.

Thank you

closed 2019-02-16 by close-issue-app[bot]

connortechnology, 2019-02-19:
@ogrebgr You submitted this PTZ script, can you comment/verify?

ogrebgr, 2019-02-19:
I am afraid that this camera is no longer with me and I cannot verify. 
However, I am noticing that in my script there is no `my $cmd = "/form/presetSet?fla`.
All commands are like: `my $cmd = "cgi/ptz_set?`
@koetj can see the original script at: https://github.com/ZoneMinder/zoneminder/blob/10531df54312f52f0f32adec3d4720c063897b62/scripts/ZoneMinder/lib/ZoneMinder/Control/Floureon.pm

Probably he tried to reveng the protocol sniffing at the http traffic of the windows tool. However I was unable to make it work that way (with /form/).

Storing preset is done with:
`my $cmd = "cgi/ptz_set?Channel=1&Group=PTZCtrlInfo&PresetNumber=1&Preset=0";`

I can send him the PDF that I have received from Floureon if he needs it. 

Also please note that there are several revisions of 1080P camera.

windowfinn, 2021-10-11:
I appreciate that this ticket is closed but I am experiencing the same issue with my Floureon 1080p. If I amend the Floureon.pm as described in the original ticket - it works. Also the existing methods do not seem to change preset value being used, it pulls it from the request, but then does not use it.

```
Set
   #my $cmd = "cgi/ptz_set?Channel=1&Group=PTZCtrlInfo&PresetNumber=1&Preset=0";
    my $cmd = "form/presetSet?flag=3&existFlag=1&language=cn&presetNum=$preset";

Get

   #my $cmd = "cgi/ptz_set?Channel=1&Group=PTZCtrlInfo&PresetNumber=1&Preset=1";
    my $cmd = "form/presetSet?flag=4&existFlag=1&language=cn&presetNum=$preset";
```
Is there a way for me to submit my findings?