ZoneMinder/zoneminder #2532: Sorting broken with more than 100 cameras

Disassembler0, 2019-02-19:
**Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04
- Browser name and version: Firefox 65, Chromium 71, Internet Explorer 11

**Describe the bug**
With more than 100 cameras, the console table layout gets split every 100 rows. This makes the column widths different, but more importantly it breaks sorting (element in rightmost column) as the element has "ui-sortable-handle" class only in the first 100 rows.

As a temporary workaround, we have raised the break limit to 1000 on this line:
https://github.com/ZoneMinder/zoneminder/blob/4cd3a93e960e86471a39bc299369d9f5480d89c1/web/skins/classic/views/console.php#L235

**To Reproduce**
Steps to reproduce the behavior:
1. Have more than 100 cameras in database
2. Open index page (default console view)
3. Scroll down to 101th camera (first below table split)
4. Try to reorder it

**Expected behavior**
Either no table splits after 100 lines or sortable rows even after table split.

Disassembler0, 2019-11-30:
I think this can be kept closed as it's a duplicate of #2533.

connortechnology, 2019-11-30:
yup

closed 2019-11-30 by connortechnology