ZoneMinder/zoneminder #2539: Extra events showing up with bogus data and no content

a-bali, 2019-02-20:
**Describe Your Environment**
- Latest version pulled from github
- from source in docker
- ubuntu xenial docker image

**Describe the bug**
I am getting extra events in the event list without no real data and with a negative timespan. See attached screenshot. Besides this, I am getting normal events perfectly too. This happens for all of my cameras.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Event list from anywhere
2. See extra events in event list
[image]

**Expected behavior**
These events do not make any sense, they don't exist on disk and should not appear.

**Debug Logs**
There is nothing relevant in the logs (apart from zmfilter failing to process these actually non-existent events).

connortechnology, 2019-02-22:
The (r) tells use that it has been re-created by zmaudit.  We would need debug logs from zmaudit.

Please note that you really don't need to run zmaudit. It is putting a lot of load on your server.

a-bali, 2019-02-27:
This still happens on a fresh install. What does the * represent?

connortechnology, 2019-02-28:
The * means archived. The r means that it was recovered by zmaudit.

Please note that I have just added support to the web ui to support cleaning logs by line count so if you run master, you will not need zmaudit anymore.

connortechnology, 2019-02-28:
I should further note that for performance reasons, deleting from logs by line count is not efficient.  First we have to count how many lines there are, and that will hit every row in the db.  So be sure to keep the line count low.

closed 2019-05-06 by stale[bot]