ZoneMinder/zoneminder #2540: 500 Error on "ADD NEW SERVER" page

brwainer, 2019-02-22:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: installed from RPMFusion following exact instructions
- Full name and version of OS: CentOS Linux release 7.6.1810 (Core) - Kernel 3.10.0-957.5.1.el7.x86_64

**Describe the bug**
The "ADD NEW SERVER" page for a multi-server install results in an HTTP/500. This is on a completely new multi-install setup that is unrelated to prior installs. That being said, this page also gives an HTTP/500 on existing single-server installs.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Options > Servers'
2. Click on 'ADD NEW SERVER' ( https://{ip}/zm/index.php?view=server&id=0&popup=1 )
3. See error (if in Chromium - Firefox just shows a blank page)

**Expected behavior**
Add new server page works.

**Debug Logs**
This is from /var/log/httpd/ssl_error_log and there is a new entry every time the page is accessed:
```
[Fri Feb 22 00:53:06.617099 2019] [:error] [pid 6120] [client 192.168.200.3:53370] PHP Fatal error:  Can't use method return value in write context in /usr/share/zoneminder/www/skins/classic/views/server.php on line 42, referer: https://192.168.200.30/zm/index.php?view=options&tab=servers
```

connortechnology, 2019-02-22:
I have merged a fix to the release-1.32 branch.  Master did not suffer from this issue.

closed 2019-02-22 by connortechnology

G8nd0lf, 2019-07-19:
I am using the RPM from RPMFusion that comes with zoneminder-1.32.3-1.el7.x86_64
If this was merged back in Feb when will it be available to use.
Do we have a work around until it is merged into the RPMFusion RPM?
Any manual steps we can do, for those using the RPM?

Thanks in advance for any advise

dscheive, 2019-11-24:
I'm getting this same error on centos 7 with RPMFusion, is there any fix?

commit 9a2dd06e (2019-02-22): Fix use of empty which isn't supported in old php.  Remove the code entirely as I think it was just cutnpasted from somewhere else.  We don't care if it is a new server or not in the validate code.
Fixes #2540