ZoneMinder/zoneminder #2546: /zm path shows nothing (blank page)

TSEngineer, 2019-03-09:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]: Release version
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]: PPA
- Full name and version of OS: Ubuntu 18.04.2 LTS (GNU/Linux 4.15.0-46-generic x86_64)
- Browser name and version (if this is an issue with the web interface): Firefox 65.0.2 (x64)

**Describe the bug**
When I am trying to access /zm path, I just see the blank page

**To Reproduce**
Steps to reproduce the behavior:
1. Go to (IP)/zm
2. Now see, there is nothing

**Expected behavior**
It must show 

**Debug Logs**
I didn't find any logs

connortechnology, 2019-03-11:
The relevant logs will be the apache logs probably in /var/log/http/* or something similar.  The blank page error occurs when something in php land has crashed.  We can not offer useful advice with out the logs containing the error.

closed 2019-05-17 by stale[bot]

Scobber, 2020-09-02:
Sorry to reopen this old old post but its one of the top google results still.

I recently had this issue, and for anyone seeking a possible solution. After I upgraded ZM to 1.34.19 (sometime around June/July). I had intermittent issues, old browsers worked fine as in connected once, but new browsers never connected before showed the white page.

It was because somewhere along the line the skin param was not set to a default. Older browsers worked because they retained their session / cookie zmSkin.

Anyway setting the skin in the URI fixed the computer I was working on to get into settings to set the default.

try this it might resolve your issue quickly
http://uri.to/zoneminder/zm/?skin=classic

Full writeup
[https://thescobber.com/2020/08/21/zoneminder-white-screen/](https://thescobber.com/2020/08/21/zoneminder-white-screen/)