ZoneMinder/zoneminder #2547: API no longer uses ZMSESSID session cookies

mnoorenberghe, 2019-03-11:
**Describe Your Environment**
- Version of ZoneMinder: zoneminder_1.33.3~20190310225825-bionic_amd64.deb
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04.2
- Browser name and version (if this is an issue with the web interface)

**Describe the bug**
If I log in (via the login page which was traditionally the only way to login for API access), my login session isn't used to grant me access to the API.

I believe this was caused by #2487 at https://github.com/ZoneMinder/zoneminder/commit/2b90bf15a6d357819b16c56dcf24dec2baf5892d#diff-97fedc73ad0b1884b6f8d2c1337674a8R217

Changing `session_start();` to the new `zm_session_start();` fixes the issue but I wonder if there is a reason why that wasn't done in the first place? I'm also not sure if the `session_write_close();` should also change.

**To Reproduce**
Steps to reproduce the behavior:
1. Successfully login to the web interface
2. Visit `/zm/api/host/getVersion.json`

**Expected behavior**
Returned:
```js
{
    "version": "1.33.3",
    "apiversion": "1.0"
}
```

**Actual behaviour**
```js
{
    "success": false,
    "data": {
        "name": "Not Authenticated",
        "message": "Not Authenticated",
        "url": "\/zm\/api\/host\/getVersion.json",
        "exception": {
            "class": "UnauthorizedException",
            "code": 401,
            "message": "Not Authenticated",
            "trace": [
                "#0 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Event\/CakeEventManager.php(243): AppController->beforeFilter(Object(CakeEvent))",
                "#1 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Controller\/Controller.php(677): CakeEventManager->dispatch(Object(CakeEvent))",
                "#2 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Routing\/Dispatcher.php(189): Controller->startupProcess()",
                "#3 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Routing\/Dispatcher.php(167): Dispatcher->_invoke(Object(HostController), Object(CakeRequest))",
                "#4 \/usr\/share\/zoneminder\/www\/api\/app\/webroot\/index.php(107): Dispatcher->dispatch(Object(CakeRequest), Object(CakeResponse))",
                "#5 {main}"
            ]
        },
        "queryLog": []
    }
}
```

**Debug Logs**
Nothing interesting in the logs but see trace above.

The `PHPSESSID` cookie gets set instead of `ZMSESSID`.

connortechnology, 2019-03-11:
As the author of the changes, let me state that Any start_session should probably be changed to zm_start_session.   sadly the php built in functions just don't do enough.

mnoorenberghe, 2019-03-11:
OK, thanks @connortechnology, I can make that change. What about my question about `session_write_close`? 
> I'm also not sure if the session_write_close(); should also change.

mnoorenberghe, 2019-03-11:
Re-opening to wait on the answer to the above comment.

connortechnology, 2019-03-18:
I don't think session_write_close needs to change.

commit 3c31dd63 (2019-03-11): Use zm_session_start() for API auth. Fixes #2547
closed 2019-03-18 by connortechnology