ZoneMinder/zoneminder #2555: zmcontrol.pl processes left running after stop/restart of zoneminder

cnighswonger, 2019-03-15:
**Describe Your Environment**
- Version of ZoneMinder: 1.30.4
- How you installed ZoneMinder: PPA
- Full name and version of OS: Linux xyz 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux

**If the issue concerns a camera**
Control module: Dahua.pm

**Describe the bug**
Restarting ZoneMinder does not result in termination of zmcontrol processes.

**To Reproduce**
Steps to reproduce the behavior:
1. Exercise any camera control function.
2. List zmcontrol.pl processes, note the PIDs
3. Stop/restart ZoneMinder
4. List zmcontrol.pl processes and compare PIDs with the list from step two.

**Expected behavior**
All processes spawned by ZoneMinder should terminate upon a stop or restart.

**Debug Logs**
```
root@dvr# ps -ef | grep zmcontrol
www-data  5396     1  4 13:11 ?        00:00:00 /usr/bin/zmcontrol.pl --id 2
root      5398 15318  0 13:11 pts/1    00:00:00 grep zmcontrol
root@dvr# service zoneminder restart
root@dvr# ps -ef | grep zmcontrol
www-data  5396     1  0 13:11 ?        00:00:00 /usr/bin/zmcontrol.pl --id 2
root      5594 15318  0 13:12 pts/1    00:00:00 grep zmcontrol
```

connortechnology, 2019-03-18:
This shouldn't be too difficult to fix, just need to use zmdc.pl to startup zmcontrol.pl

cnighswonger, 2019-03-18:
Thanks for the tip. I'll dig around and see about fixing it.

PR #2563 (2019-04-01): daemonize zmcontrol
Fixes #2555 

Previously zmcontrol was started on the fly by the web php when a ptz command was sent. This created a process not managed by zmpkg/zmdc. Thus when zoneminder was stopped the process was left running.

This PR attempts to properly daemonize zmcontrol. If a monitor is controllable, zmpkg will spawn a zmcontrol process during startup for that monitor. Zmcontrol has been added to zmdc's list of daemons, which results in proper shutdown of the process when ZoneMinder is stopped.

Accomplishing this required modifying zmcontrol to startup w/o a ptz command. 
I am m [... cut]
closed 2019-04-01 by connortechnology