ZoneMinder/zoneminder #2556: 1.32.3 Restarting capture daemon - shared data not valid

dac2020, 2019-03-15:
Ubuntu 18.04 server
ZoneMinder 1.32.3
Installed using the ubuntu script found at;
https://wiki.zoneminder.com/Ubuntu_Server_18.04_64-bit_with_Zoneminder_1.32.x_the_easy_way

New install, two IP cameras. One a very basic IP camera and the one not working is as follows;
INS-HD360A-5MP-4X (bought on ebay)
*PTZ IP Bullet Camera Outdoor / Indoor
*ONVIF2.4;Support P2P function
*Support H.264 /H.265 compression
*Pan/Tilt/Zoom, A-B scan, tour available
* Compatible with HIKVISION ,DAHUA, XM NVR
*Compatible with Milestone, NUUO and AXXON platform

Specification
Camera
Sensor: 1/3" CMOS 5MP AR0521
Output pixel: 2592x1944@15fps / 2592x1520@22fps
Lens: 4X f=2.8~12MM
BLC: Support
AGC: Auto/manual
WB: Auto/manual/indoor/outdoor
S/N Ratio>50dB

I'm sorry, I'm not sure what information to provide so I've got the above and the following. I'll include some logging which I hope could help. It seems as if the camera becomes disconnected from ZM but the weird thing is that I am monitoring it from another machine and never see any drops. I have live panning so can see if it misses anything.

However, in the logging I keep seeing errors and the monitor keeps going red. I've been searching the net since yesterday and find a ton of information but nothing which helps me to confirm what the problem is or what to do. 

Figured I should post it here just in case it's something that dev's aren't aware of. I've read many posts saying this was fixed but since I can't quite confirm the problem, I'm not sure what to fix.

Mar 15 21:22:34 zoneminder zmc_m7[8180]: ERR [zmc_m7] [Unable to send packet at frame 201: Resource temporarily unavailable, continuing]
Mar 15 21:22:35 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:39700 - Capturing at 25.00 fps, capturing bandwidth 7842487bytes/sec]
Mar 15 21:22:38 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:39800 - Capturing at 33.33 fps, capturing bandwidth 10424337bytes/sec]
Mar 15 21:22:42 zoneminder zmc_m7[8180]: ERR [zmc_m7] [Unable to send packet at frame 212: Resource temporarily unavailable, continuing]
Mar 15 21:22:42 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:39900 - Capturing at 25.00 fps, capturing bandwidth 7816403bytes/sec]
Mar 15 21:22:42 zoneminder zmc_m7[8180]: ERR [zmc_m7] [Unable to send packet at frame 212: Resource temporarily unavailable, continuing]
Mar 15 21:22:42 zoneminder zmc_m7[8180]: message repeated 4 times: [ ERR [zmc_m7] [Unable to send packet at frame 212: Resource temporarily unavailable, continuing]]
Mar 15 21:22:45 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40000 - Capturing at 33.33 fps, capturing bandwidth 10432579bytes/sec]
Mar 15 21:22:47 zoneminder zmwatch[1448]: INF [Restarting capture daemon for ptz back-yrd, time since last capture 6 seconds (1552684967-1552684961)]
Mar 15 21:22:48 zoneminder zmdc[1373]: INF ['zmc -m 7' sending stop to pid 8180 at 19/03/15 21:22:48]
Mar 15 21:22:48 zoneminder zmc_m7[8180]: INF [zmc_m7] [Got signal 15 (Terminated), exiting]
Mar 15 21:22:49 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40100 - Capturing at 25.00 fps, capturing bandwidth 7823360bytes/sec]
Mar 15 21:22:49 zoneminder zmdc[1373]: INF ['zmc -m 7' exited normally]
Mar 15 21:22:52 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40200 - Capturing at 33.33 fps, capturing bandwidth 10420224bytes/sec]
Mar 15 21:22:56 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40300 - Capturing at 25.00 fps, capturing bandwidth 7817733bytes/sec]
Mar 15 21:22:57 zoneminder zmfilter_1[1432]: INF [Checking filter PurgeWhenFull delete returned 0 events ]
Mar 15 21:22:57 zoneminder zmfilter_2[1438]: INF [Checking filter Update DiskSpace update disk space returned 0 events ]
Mar 15 21:22:58 zoneminder zmwatch[1448]: ERR [Memory map file '/dev/shm/zm.mmap.7' does not exist.  zmc might not be running.]
Mar 15 21:22:58 zoneminder zmwatch[1448]: INF [Restarting capture daemon for ptz back-yrd, shared data not valid]
Mar 15 21:22:59 zoneminder zmdc[1373]: INF [Command 'zmc -m 7' removed from pending list at 19/03/15 21:22:59]
Mar 15 21:22:59 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40400 - Capturing at 33.33 fps, capturing bandwidth 10418786bytes/sec]
Mar 15 21:23:03 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40500 - Capturing at 25.00 fps, capturing bandwidth 7814898bytes/sec]
Mar 15 21:23:06 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40600 - Capturing at 33.33 fps, capturing bandwidth 10442316bytes/sec]
Mar 15 21:23:09 zoneminder zmwatch[1448]: ERR [Memory map file '/dev/shm/zm.mmap.7' does not exist.  zmc might not be running.]
Mar 15 21:23:09 zoneminder zmwatch[1448]: INF [Restarting capture daemon for ptz back-yrd, shared data not valid]
Mar 15 21:23:10 zoneminder zmdc[1373]: WAR [Can't find process with command of 'zmc -m 7']
Mar 15 21:23:10 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40700 - Capturing at 25.00 fps, capturing bandwidth 7817322bytes/sec]
Mar 15 21:23:10 zoneminder zmdc[1373]: INF ['zmc -m 7' starting at 19/03/15 21:23:10, pid = 8192]
Mar 15 21:23:10 zoneminder zmdc[8192]: INF ['zmc -m 7' started at 19/03/15 21:23:10]
Mar 15 21:23:11 zoneminder zmc_m7[8192]: INF [zmc_m7] [Starting Capture version 1.32.3]
Mar 15 21:23:11 zoneminder zmc_m7[8192]: INF [zmc_m7] [Priming capture from rtsp://"admin":"11112222"@192.168.1.101:554/1/h264major]
Mar 15 21:23:12 zoneminder zmc_m7[8192]: INF [zmc_m7] [Stream open rtsp://"admin":"11112222"@192.168.1.101:554/1/h264major, parsing streams...]
Mar 15 21:23:13 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40800 - Capturing at 33.33 fps, capturing bandwidth 10426136bytes/sec]
Mar 15 21:23:13 zoneminder zmc_m7[8192]: INF [zmc_m7] [ptz back-yrd: images:1 - Capturing at 50.00 fps, capturing bandwidth 109792bytes/sec]
Mar 15 21:23:17 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40900 - Capturing at 25.00 fps, capturing bandwidth 7818840bytes/sec]
Mar 15 21:23:19 zoneminder zmc_m7[8192]: ERR [zmc_m7] [Unable to send packet at frame 71: Resource temporarily unavailable, continuing]

closed 2019-03-15 by close-issue-app[bot]

dac2020, 2019-03-15:
You closed the issue because I didn't follow a standard? Ever hear of editing? I could edit it. I took the time to help if I can. Not a nice way to treat people.

connortechnology, 2019-03-16:
Closing is automatic.  

Anyways, do you have values set for max fps and alarm fps in zm?  If so, clear them.  In 1.30 they were needed but that is no longer the case and can cause ZM to not keep up with the stream.

dac2020, 2019-03-16:
Thanks for the information. 
Yes, both were set to 30.0 by default. I made the change you suggested and this seems to have stopped those errors. In fact, this seems to have stopped the shared memory errors too. 

There are other errors left such as signal loss which make no sense as I am monitoring those same cameras from other locations without problems. In the logs I keep seeing something restarting, failures etc. 

I'll post the last bits of errors I am seeing because I have no way of knowing if this is a bug or just a configuration problem. Searching the Internet for the last two days seems to imply it could be either and I've come across github issues that seem to talk about similar errors and being called 'bug'.

Mar 16 15:51:19 zoneminder zmc_m2[19007]: ERR [zmc_m2] [Bogus number of lines return from query, 0 returned for query SELECT Id, Name, ServerId, StorageId, Type, Function+0, Enabled, LinkedMonitors, AnalysisFPSLimit, AnalysisUpdateDelay, MaxFPS, AlarmMaxFPS,Device, Channel, Format, V4LMultiBuffer, V4LCapturesPerFrame, Protocol, Method, Options, User, Pass, Host, Port, Path, Width, Height, Colours, Palette, Orientation+0, Deinterlacing, RTSPDescribe, SaveJPEGs, VideoWriter, EncoderParameters, RecordAudio, Brightness, Contrast, Hue, Colour, EventPrefix, LabelFormat, LabelX, LabelY, LabelSize,ImageBufferCount, WarmupCount, PreEventCount, PostEventCount, StreamReplayBuffer, AlarmFrameCount, SectionLength, FrameSkip, MotionFrameSkip, FPSReportInterval, RefBlendPerc, AlarmRefBlendPerc, TrackMotion, Exif, SignalCheckPoints, SignalCheckColour FROM Monitors WHERE Id=2.]

In fact, it's not worth posting the rest of the errors. I'll try posting the rest in the forums.

Thanks.