ZoneMinder/zoneminder #2559: Unable to authenticate user (zms.cpp 212) - Cycle, Montage & Montage Review tabs affected

dwrobel, 2019-03-19:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
```
$ rpm -qv zoneminder
zoneminder-1.32.3-1.fc28.x86_64
```
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
Official fedora repo.
- Full name and version of OS
```
$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 28 (Twenty Eight)
Release:	28
Codename:	TwentyEight
```
- Browser name and version (if this is an issue with the web interface)
```
rpm -qv firefox
firefox-65.0.2-1.fc28.x86_64
```
- Relevant part of the config:
- OPT_USE_AUTH - yes
- AUTH_TYPE - builtin
- AUTH_RELAY - hashed
- AUTH_HASH_IPS - yes
- AUTH_HASH_LOGINS - yes
- OPT_USE_GOOG_RECAPTCHA - no
- Apache HTTP server

**Describe the bug**
On Cycle, Montage & Montage Review tabs I don't see a video. Interestingly all other tabs _(e.g. Console, Options, Log)_ works fine _(display relevant information)_. On the Log tab the error is:
```

2019-03-19 19:09:45 | zms_m2 |   | 22074 | ERR |  Unable to authenticate use | zms.cpp | 212
-- | -- | -- | -- | -- | -- | -- | --
2019-03-19 19:09:36 | zms_m1 |   | 22067 | ERR | Unable to authenticate user | zms.cpp | 212
```
but while I'm accessing any tabs the UI [doesn't ask me for any authentication](https://dwrobel.fedorapeople.org/tmp/zoneminder/Screenshot%20from%202019-03-19%2019-22-44.png).

**To Reproduce**
The phenomenon seems to be reproducible every time I'm following the scenario described below:
1. Connect to the ZM being out of home.
2. When back in home _(usually after few hours)_ refresh the Firefox tab used to connect to ZM, then observe the situation as described beforehand.

**Expected behavior**
Either every ZM tab should work _(present it's data correctly)_ or none of them should provide any access to data prior asking user to re-authenticate.

**Workaround**
Logout/login and Cycle, Montage & Montage Review tabs works again _(display video)_.

knight-of-ni, 2019-03-31:
Are you performing each test from the same computer? If so, this is caused by AUTH_HASH_IPS being enabled. If the IP address of the client pc changes, then that breaks the hash that is generated. The reason zm appears to partially break, and not completely, is the authentication hash is only used when a stream is being sent to the client. It is not used anywhere else.

I agree certainly that this could be handled better. In the meantime, turning off AUTH_HASH_IPS should allow your client pc to work w/o issue after moving to a different ip network.

Giulio2333, 2019-04-01:
I got something similar, but I can't see live view in any way.
Not sure is the same thing though, so I opened a new issue: #2565

dwrobel, 2019-05-31:
> Are you performing each test from the same computer? 

Yes, it's the same machine.

>If so, this is caused by AUTH_HASH_IPS being enabled.

Disabling it helped.

Thank you for the hint.

tahphallus, 2019-06-24:
I have also tried disabling AUTH_HASH_IPS but that dit not work.
Only setting AUTH_RELAY to NONE helped

knight-of-ni, 2019-06-24:
@tahphallus yours is a different issue

closed 2019-08-30 by stale[bot]