ZoneMinder/zoneminder #2564: Zoneminder Corrupts event videos

RobertDeRose, 2019-04-01:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: deb http://ppa.launchpad.net/iconnor/zoneminder-1.32/ubuntu bionic main       
- Full name and version of OS:
NAME="Ubuntu"                                                                 
VERSION="18.04.2 LTS (Bionic Beaver)"  
- Browser name and version:
Chrome - 73.0.3683.86 (Official Build) (64-bit)
Safari -  12.1 (14607.1.40.1.4)

**If the issue concerns a camera**
- Make and Model: HJT IPC-6200-B1
- frame rate: 20
- resolution: 800 x 448 / 1920 x 1080
- ZoneMinder Source Type: h264 via ffmpeg camera pass through

**Describe the bug**
Events are capture and when I click on the event to view it I get this:

[image]

Also, if I click the "DOWNLOAD MP4" the video is unplayable, however if I download the video via SFTP the video is playable.

**To Reproduce**
For me this was a fresh install and simply has never worked.
I previously had 1.30 installed, but did a fresh install when all cameras stopped capturing after upgrading.

**Expected behavior**
Viewing an event should work and ZoneMinder should not corrupt the video.

**Debug Logs**
Nothing in the logs point to this issue that I can find. But give me something to look for and I'll send them along

RobertDeRose, 2019-04-01:
Here are the videos

```
sha256sum 135-video_sftp.mp4
016cc95999525ce47f0255dc03b062e6a45b13b42bdc76c897b801b660500b89  135-video_sftp.mp4
sha256sum 135-video_download_butoon.mp4
02fb2acd1ab0a30b4675c344a2d5cebb9edb9f1ef074666cb863097f72e8b262  135-video_download_butoon.mp4
```

https://www.dropbox.com/sh/i5lqt6o4tgigvog/AADPwCn1iGNL8uNEMZaWhn3Ia?dl=0

file 135-video_sftp.mp4
135-video_sftp.mp4: ISO Media, MP4 Base Media v1 [IS0 14496-12:2003]

file 135-video_download_butoon.mp4
135-video_download_butoon.mp4: data

RobertDeRose, 2019-04-01:
doing head on the corrupt file, I see the issue, but not sure how to fix.  The corrupt version contains this

```bash
<br />
<b>Notice</b>:  ob_clean(): failed to delete buffer. No buffer to delete in <b>/usr/share/zoneminder/www/views/view_video.php</b> on line <b>101</b><br />
    ftypisom   isomiso2avc1mp41   free _Fmdat   )gd ≠Ñ a C@¬ Ñ;Pd”p      ¢   hÓ<∞  ñ¶e∏  ?ÛÉ∂*Ç÷g“d¡le+9;{R¥
```

RobertDeRose, 2019-04-01:
By commenting out `ob_clean` on line 101 I now get an uncorrupted video that plays, but it fails at the end exactly as the comment above the line says I would

[image]

RobertDeRose, 2019-04-01:
Looks like this is a common issue possibly.  

https://www.drupal.org/project/drupal/issues/205227

I'm not a perl guy, so no idea. I'd be glad to make a PR if someone here can confirm this is the fix they would do, but happy to have another thought.

N3X15, 2019-04-01:
Would probably be a better bet to disable E_NOTICE in the script that streams the video files, and/or redirect error messages to a logfile rather than stdout.

knight-of-ni, 2019-04-01:
The ob_clean issue is a duplicate of #2390 which was fixed in #2395. Please upgrade to the latest master branch and test.

RobertDeRose, 2019-04-01:
@knight-of-ni how does one go about updating to master when using an Ubuntu Package?

connortechnology, 2019-04-01:
Use our master ppa.

sudo add-apt-repository ppa:iconnor/zoneminder-master

closed 2019-04-21 by no-response[bot]